rstrouse/ESPSomfy-RTS #420: Errror show on main page

ionutm80, 2024-07-02:
### Hardware

ESP32-S3

### Firmware version

v2.4.5

### Application version

v2.4.5

### What happened? What did you expect to happen?

Hi, I have 2 errors shown on the main page, see print screen attached.
1. One show an exclamation mark in a red triangle for one of the blinds although it works perfectly.
2. One show that the 2nd blind has a sun sensor although I do not have such a thing.
3. 
[image]
[image]
[image]

### How to reproduce it (step by step)

```bash
1. Go to main page where each blind is shown.
```

### Logs

```bash
No line present in the log section.
```

rstrouse, 2024-07-02:
That is not an error.  You have two indicators from a Sun/Wind sensor.  The triangle is for the wind and the little sun is for the sun.

If you have these indicators erroneously on your display and you do not have a sensor, then you likely triggered these on the virtual remote page. To clear them send a sensor command from the virtual remote without any of the checkboxes checked.

ionutm80, 2024-07-02:
Yesit worked! Yeay!!! Thanks!

closed 2024-07-02 by rstrouse