rstrouse/ESPSomfy-RTS #422: Blinds pair ok but only move small amount with every press up or down

benstephenson85, 2024-07-03:
Excellent project, thank you so much. 

I successfully paired four Roman blinds and they worked well. I'm unsure why but now when I press the buttons in the Somfy RTS Web UI, the blinds only move a centimeter or so. I've tried selecting different types of blinds in the setup with no change. The Somfy remote works as expected and the blinds travel fully up or down. I have other blinds and they still work perfectly. Sorry to trouble you but not sure if this is a bug that others are experiencing too. Thanks.

rstrouse, 2024-07-03:
Try this.  Go into the virtual remote then select a blind from the dropdown.  Press and hold the Up + Down button for at least 5 seconds.   If the blind jogs, stop pressing the button.  This may switch the motor out of tilt only mode.

benstephenson85, 2024-07-03:
Thanks but although the blind does jog after holding the Up + Down button it doesn't seem to make a difference to the behaviour. I'm not sure why but when I press down on the virtual remote the blind does two steps down, when I press the down button on the page for the blind in the UI it only steps down once. The Somfy remote (Situo 5 RTS Pure II) does fully up and down to the limits. Note that there is only up and down control for a Roman blind so would there be a tilt mode for it to get stuck in?

rstrouse, 2024-07-03:
The motors are simply mechanical and the tilt mechanisms are as well.  The motor has no idea, regarding what type of mechanism it is installed into.

If while in the Virtual remote, you press the up/down button for longer does it go fully up and down?  The virtual remote is a clever little bugger where it senses whether the button remains pressed whereas the home screen reflects the defined repeats.

benstephenson85, 2024-07-03:
Yes holding the button works, it's a bit slower than usual but the blinds go all the way up and down to the programmed stops.

rstrouse, 2024-07-05:
I don't have your exact motors and I could only find this in English.

https://www.somfysystems.com/en-us/support/faq?question=what-to-do-if-your-rts-motor-only-moves-when-you-hold-up-or-down

benstephenson85, 2024-07-08:
I've managed to resolve this behaviour with motors Somfy RTS Tilt and Lift 25. Similar to your suggestion but slightly modified steps for these particular motors.Press and hold Up + Down on the virtual remote (not the Somfy remote) when the blinds are not at the stops I.e. part way open. Once the blind jogs, press and hold My + Down. This switches the mode for the remote that is sending the commands and the blinds will then fully open or close with a single press of the up or down button.

https://www.somfysystems.com/en-us/support/faq?question=what-to-do-if-your-rts-motor-only-moves-when-you-hold-up-or-down

rstrouse, 2024-07-08:
Thanks I have taken note of this for the Tilt and Lift motors.  I am going to close this for now and add it to a planned FAQ troubleshooting wiki that I am planning for things like this.

closed 2024-07-08 by rstrouse