rstrouse/ESPSomfy-RTS #424: HA UI for open/close+tilt shades is not working correctly

hajar97, 2024-07-10:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

When I use HA UI to change tilt of the shades it correctly changes tilt of the shades, but in parallel with it sets position of the shade to the same %. 

If I do that programmatically by setting tilt and position to whatever value, then everything works correctly. This seems to be HA UI issue only.

Initial state of shade:
[image]

Position of the shade after setting tilt to 50% (right column):
[image]

Expectation is that only right column goes into 50%, while left column remains where it is initially. Physically this is exactly what happens, but visually in HA UI not.

When checking state attributes in HA developer tools, their state also changes incorrectly.
Initial value of attributes is:
current_position: 50
current_tilt_position: 0

After using the UI to set the Tilt position to 50% I get:
current_position: 50
current_tilt_position: 0

Expectation is exactly the opposite, that current_tilt_position changes to 50 and current_position remains at 0.

If I use UI to set shade position (left column) instead of tilt (right column) then all seems to work correctly. Shade position gets set to the value I set it via UI and shade tilt remains where it was before. 
However, UI playback is wrong. Whatever value I set position to via UI, it visually shows that shade first gets fully opened and then set to position I set it to, instead of just gradually opening/closing to the position I set it to as it happens physically.

Tilt type of my shade is integrated, meaning it changes the tilt depending the direction of opening/closing.

### How to reproduce it (step by step)

```bash
1. Close shade fully to position 0 and tilt 0
2. Then using HA UI set shade tilt to 50%.
3. HA UI will start displaying how tilt is moving to 50%, but in parallel with it shades position also visually moving to 50%, while I expect that it should stay in place and not move.
```

rstrouse, 2024-07-10:
I'll have a look and I know that this used to work correctly in HA and it looks like the ESPSomfy RTS UI is doing the correct thing as well since the target appears to be 100 on your screenshot.

hajar97, 2024-07-10:
Yes, confirm that this definitely used to work. I don’t know if this is due to recent updates to HA or due to latest firmware update, but it definitely stopped working correctly, while it did before.

rstrouse, 2024-07-11:
I found the issue and it turns out I made a mistake with the keys.  I will have a fix shortly.

rstrouse, 2024-07-11:
I posted a new release for the HA component that should fix your issue.

hajar97, 2024-07-11:
How can I try it? I tried installing 2.4.6-pre and that didn’t fix it. Same behaviour.

rstrouse, 2024-07-11:
No this is the HA component.  v2.4.7 will show up in the next day or so in your updates.  Or if you open the HACS store it will refresh and find the update.

hajar97, 2024-07-11:
Done. Seems to have fixed the issue. Thank you.

rstrouse, 2024-07-11:
Awesome!  Closing this one out.

closed 2024-07-11 by rstrouse