rstrouse/ESPSomfy-RTS #425: Lost access to web UI

rein-10, 2024-07-11:
### Hardware

ESP32

### Firmware version

v2.4.4

### Application version

v2.4.4

### What happened? What did you expect to happen?

So I been using the esp somfy rts for the last couple months is was working great. In the last week i couldnt update it since i couldnt access the web UI it only showed the logo of esp somfy but i could still use it with home assitante. The problem now is have lost control since i cannot access the web UI and cannot use home assitant. I have tried restarting a couple times with no luck, when I ping the ip address of the esp i get a response but still cannot access the web UI. Any recomendations?

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-07-11:
Remove power from the ESP32 then power it up again.

Now make a backup of your configuration by opening a web browser and enter the following command into the browser window.

`http://<ip address>/backup`

This should download a backup file with your configuration.  You can use this if you need to reflash the onboard file.

Next run the following command.  Note this command could take a couple of minutes.  Give it some time to restart after it is done.

`http://<ip address>/downloadFirmware?ver=latest`

rein-10, 2024-07-13:
worked great thanks!

rstrouse, 2024-07-13:
Awesome.  I am going to close this out.

closed 2024-07-13 by rstrouse