rstrouse/ESPSomfy-RTS #426: Override/ignore pin restrictions

mhofer117, 2024-07-12:
### Describe you new feature you'd like

First off, thanks for this great project!

I am using an M5Stack Atom S3 and during radio transceiver setup I could not select the desired pins.
There appear to be some pin restrictions from [index.js#L2786](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/data/index.js#L2786).

Due to the restricted number of pins on the Atom S3, I want to use GPIO-39. However it is defined as input and output in pinMap so is not selectable at all.
Documentation shows that on the Atom S3 this should be available as a normal GPIO pin.
The wiring looks as follows:
[image]

I am aware that rx/tx can be combined on gdo0 or gdo2 (setting both to the same), but it would be nice to have them separate as enough GPIO pins are available.

For the moment I stored the pins using a manual REST call using GPIO-39 as SCLK pin. This seems works and I can use the device.
However I can not store the transceiver configuration again in the web-ui as it is getting reset to GPIO-1 there.

It would be nice if there is an option in the UI to ignore pin restrictions, as they seem to be different across board variants anyway (#367)

If you would consider this feature request is acceptable I could craft a PR with a possible solution.

rstrouse, 2024-07-12:
Hmmm... 39 is an external clock pin on the S3 platform.  I am surprised that this does not have ill effects but I guess it does not matter if the clock settings are the same.  I will take another look at the pin restrictions and remove those that may be questionable.   

The reason that I started restricting pins was because I had folks assigning inputs to outputs and choosing pins on ESP32 that were unusable on the devkit.  The S series chips seem to have removed restrictions on the common ones that are actually exposed although there are many pins still that will not be exposed on the board.

mhofer117, 2024-07-13:
> I am surprised that this does not have ill effects but I guess it does not matter if the clock settings are the same.

Maybe there are ill effects I am not aware of? I am not too experienced with this, but I can control the blind and also see  commands from the physical remote in the log. What would I need to look out for?

rstrouse, 2024-07-13:
I think I understand why you do not experience ill effects from this.  On the S3 GPIO39 is part of the JTAG interface but I disable that interface in the firmware since we have no use for it.  I originally thought this pin was being used for another purpose but it is not and is only an external clock pin when the JTAG is active.  In fact GPIO39-42 are hardwired to the JTAG and are available if that interface is not used.  While the atom does not expose these pins the S3 can actually use them.

If you update to the v2.4.6-pre several pins that were disabled before are now enabled while still preserving the reserved pins.

mhofer117, 2024-07-13:
I just updated to v2.4.6-pre and can now set/modify the transceiver pins as desired:
[image]

Thank you very much!

closed 2024-07-13 by mhofer117