rstrouse/ESPSomfy-RTS #427: Utilise any 433MHz devices others than shades?

rgr101, 2024-07-12:
### Describe you new feature you'd like

Hi, 
I'm wondering if I could also also use the ESPSomfy for detecting/controlling other devices for complete different purposes, such as an ELRO AB440 remote or an ELRO plug etc. (of course using a second device working on 433.92)? I'm not aware of the protocol they're using, but it seems to be a quite basic and common static protocol (with fixed house codes defined by dip switches and without rolling codes) ...
I still have a few of these remotes lying around and find them pretty handy, so I would like to continue using them with home assistant... (I saw an ESPHome adoption for the Sonoff RF Bridge, but I would prefer your device if possible as it is so great and moreover I'm already used to it ...😇)

dc911x, 2024-07-12:
Many other utilisations are already available, see screenshot. 

[image]

rstrouse, 2024-07-12:
If you have tuned the radio to 433.92 and have not seen frames in the log (any frames valid or not) then this is probably not possible.  There are a lot of solutions out there that try to be a swiss army knife for every possible device but they make you use the spring loaded scissors to cut leather.  They work... but not very well.  

If it is an RTx protocol as indicated by frames in the log then the dry contact functions will likely work.  If there are no frames logged then decoding it is probably a chore for someone else since I do not have any of these devices.

rgr101, 2024-07-12:
didn't try that so far (as my currently only available device is in production for moving my somfy shades VERY WELL!! 😇👍)
But yea - you're absolutely right...