rstrouse/ESPSomfy-RTS #428: Lots of unhandled exception `IntegerDivideByZero`

edrose, 2024-07-14:
### Hardware

ESP32

### Firmware version

v2.4.5

### Application version

v2.4.5

### What happened? What did you expect to happen?

I'm having massive issues trying to get this software working reliably, and I'm constantly getting an `IntegerDivideByZero` exception which causes the chip to reset.

I've tried this on v2.4.4, v2.4.5, and v2.4.6 and I'm getting the same behaviour.

I can't find a pattern to the crashes. Sometimes it works really reliably, but then if I pull the power and restart it stops working again.

Listed below is one of the ways in which I've been able to reproduce the problem, but in the past it's also ended up in a complete boot loop. Now I can't even backup the config without it crashing, so I assume something in my config file is causing this.

Config is attached below.

```
24, 76,   29,  3,  276, 12,  200,  5,   77,  1,   77,  181,   74,E37A1C   
  1,Living Room         ,  1
  2,Bedroom 1           ,  2
  3,Bedroom 2           ,  3
  1,true ,  0,    505085,Living Room 1       ,  0,  0, 56,     44000,     44000,      7000,  100,   5298149,   5494757,   5953509,   6281189,         0,         0,         0,   38,  0,    97.50000,    -1.00000,     0.00000,     0.00000,false,false,  1,  1,  0,  0,  0,  0,  1
  2,true ,  0,    505086,Living Room 2 Door  ,  0,  0, 56,     44000,     44000,      7000,  100,   5363685,   5494757,   5953509,   6215653,   6281189,         0,         0,   14,  0,    97.50000,    -1.00000,     0.00000,     0.00000,false,false,  1,  2,  0,  0,  0,  0,  1
  3,true ,  0,    505087,Living Room 3       ,  0,  0, 56,     44000,     44000,      7000,  100,   5429221,   5494757,   5953509,   6281189,         0,         0,         0,   21,  0,    97.50000,    -1.00000,     0.00000,     0.00000,false,false,  1,  3,  0,  0,  0,  0,  1
  4,true ,  0,    505088,Living Room 4       ,  0,  0, 56,     44000,     44000,      7000,  100,   5560293,   5887973,   5953509,   6281189,         0,         0,         0,   29,  0,    97.50000,    -1.00000,     0.00000,     0.00000,false,false,  1,  4,  0,  0,  0,  0,  1
  5,true ,  0,    505089,Living Room 5       ,  0,  0, 56,     44000,     44000,      7000,  100,   5625829,   5887973,   5953509,   6281189,         0,         0,         0,   36,  0,    97.50000,    -1.00000,     0.00000,     0.00000,false,false,  1,  5,  0,  0,  0,  0,  1
  6,true ,  0,    505090,Living Room 6       ,  0,  0, 56,     44000,     44000,      7000,  100,   5691365,   5887973,   5953509,   6281189,         0,         0,         0,   26,  0,    97.50000,    -1.00000,     0.00000,     0.00000,false,false,  1,  6,  0,  0,  0,  0,  1
  7,true ,  0,    505091,Living Room 7       ,  0,  0, 56,     44000,     44000,      7000,  100,   5756901,         0,         0,         0,         0,         0,         0,    2,  0,    97.50000,    -1.00000,   100.00000,     0.00000,false,false,  1,  7,  0,  0,  0,  0,  1
  8,true ,  0,    505092,Living Room 8 Corner,  0,  0, 56,     44000,     44000,      7000,  100,   5822437,   5953509,   6281189,         0,         0,         0,         0,   11,  0,    97.50000,    -1.00000,     0.00000,     0.00000,false,false,  1,  8,  0,  0,  0,  0,  1
  9,true ,  0,    505096,Bedroom 1 Left      ,  0,  0, 56,     24000,     24000,      7000,  100,   7491389,   7753533,   6150117,   6281189,         0,         0,         0,    9,  0,    75.86250,    -1.00000,     0.00000,     0.00000,false,false,  1,  9,  0,  0,  0,  0,  2
 10,true ,  0,    505097,Bedroom 1 Right     ,  0,  0, 56,     24000,     24000,      7000,  100,   7556925,   7753533,   6150117,   6281189,         0,         0,         0,    9,  0,    75.80417,    -1.00000,     0.00000,     0.00000,false,false,  1, 10,  0,  0,  0,  0,  2
 11,true ,  0,    505099,Bedroom 2 Left      ,  0,  0, 56,     24000,     24000,      7000,  100,   9332029,   9594173,   6084581,   6215653,   6281189,         0,         0,    9,  0,    71.01250,    -1.00000,     0.00000,     0.00000,false,false,  1, 11,  0,  0,  0,  0,  3
 12,true ,  0,    505100,Bedroom 2 Right     ,  0,  0, 56,     24000,     24000,      7000,  100,   9397565,   9594173,   6084581,   6281189,         0,         0,         0,   11,  0,    71.00000,    -1.00000,     0.00000,     0.00000,false,false,  1, 12,  0,  0,  0,  0,  3
  1,  0,    505093,Balcony             ,  0, 56,  1,  2,  3,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  2,  2,false,  1,   23
  2,  0,    505094,Park                ,  0, 56,  4,  5,  6,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  2,  1,false,  1,   13
  3,  0,    505095,Living Room         ,  0, 56,  1,  2,  3,  4,  5,  6,  8,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  2,  0,false,  0,   31
  4,  0,    505098,Bedroom 1           ,  0, 56,  9, 10,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  2,  3,false,  0,   10
  5,  0,    505101,Bedroom 2           ,  0, 56, 11, 12,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  0,  2,  4,false,  0,   18
         0,         0,         0,         0,         0,         0,         0
"v2.4.5","ESPSomfyRTS","pool.ntp.org","GMT0BST,M3.5.0/1,M10.5.0",true ,false
  1,false,"192.168.16.6","192.168.16.1","255.255.255.0","192.168.16.1","0.0.0.0","MQTT","homeassistant.local", 1883,true ,"somfyrts","homeassistant",  0,  0,  0,   1,   0,   0,   0
true ,  0, 56, 18,  5, 23, 19, 13, 12,   433.410,    99.97,    47.60,  10

```

### How to reproduce it (step by step)

```bash
One of the ways I can currently reproduce it is below.

1. Select blind down
2. Blind starts moving
3. Select blind up
4. Blind doesn't move, ESP32 crashes

Looking at the logs, the rolling code for the up command is '1', whereas the rolling code for the down command was '39', so something has gone very wrong there. After the reboot, the rolling code for the up command reverts to '39', indicating that the rolling code was not saved (but at least it's not '1' anymore!)
```

### Logs

```bash
Sending Shade Command
[... 161 lines cut ...]

MQTT Subscribed to:somfyrts/groups/+/windy/set

MQTT Startup Completed

Timing Net: 697ms
Socket [0] Connected from 192.168.16.5 url: /
Initializing Socket Client 0
Socket [1] Connected from 192.168.16.11 url: /
Initializing Socket Client 1
Max Heap: 106484

Free Heap: 152168

Min Heap: 137456

Timing WebServer: 629ms
Sending Shade Command

Received:{"shadeId":4,"command":"up"}

CMD:Up ADDR:505088 RCODE:39 REPEAT:1

Enabled receive on Pin #12 Timing: 1
```

rstrouse, 2024-07-14:
@edrose I loaded up your backup file and attempted to duplicate the divide by zero exception without success.  I performed the actions as you have outlined above with the firmware you have installed.

I noticed you are using MQTT discovery.  Are you using HomeAssistant then you do not need or want MQTT since the plugin will handle all of your shades way more efficiently than MQTT.  Anyhow please uncheck the enable MQTT client on the MQTT tab to see if the issue goes away.  This will allow me to see if there is something that is feeding back through the MQTT channel.

Then please install the current release of v2.4.6-pre.   This has some additional checks in it to ensure the MQTT connection is timed appropriately.  I would appreciate the exception dump like you provided above if it is still happening.   I have a sneaking suspicion that it is not really a divide by zero exception but rather the watchdog getting triggered from a feedback loop.

edrose, 2024-07-14:
I think I've solved it. I'm running on a custom PCB and I'd forgotten to add enough capacitance on the reset pin.

My best guess is that the ESP32 booted up and mounted the filesystem before the flash memory was fully initialised, so read out a zero while mounting the filesystem. That zero got saved in memory and not used until the filesystem needed to do something, at which point it was used in a division and crashed everything.

I'm guessing if you look at the address pointed to by the program counter in the core dump it will point to a function in LittleFS? I don't know which versions of libraries the firmware was built with so I can't replicate the build to find out myself.

I'll close this for now and reopen it if the issue reappears.

closed 2024-07-14 by edrose

rstrouse, 2024-07-14:
In fact it does and what you are saying makes sense.   I was beginning to think we were blowing the stack.