rstrouse/ESPSomfy-RTS #429: MQTT not connecting

passibe15, 2024-07-14:
### Hardware

ESP32

### Firmware version

v2.4.6 pre-release

### Application version

v2.4.6 pre-release

### What happened? What did you expect to happen?

Because I had some WiFi-reconnection issues, I upgraded to v2.4.6 pre-release. The WiFi issues are now solved (thanks!) but immediately after the upgrade, the ESP does not connect to MQTT anymore.

I was confused at first, thought it might be a configuration error or something, but I didn't change anything on my MQTT server.
I logged the traffic and saw that there aren't even any MQTT packets being sent out. Rebooting did not help.
Clicking on "Save MQTT settings" does also not help. I can see that the HTTP request from my browser to the `/connectmqtt` endpoint is being made and it's responded with `200 OK` and the correct JSON data, but that's it, nothing else happens.
During all that time the only network traffic coming from the ESP is on port 80 and 8080, all going to my machine. No traffic to the MQTT server at all.

Downgrading to v2.4.5 fixes the issue, a connection to the MQTT server is made immediately after uploading `SomfyController.ino.esp32.bin`.

Thanks!

### How to reproduce it (step by step)

```bash
1. Upgrade to v2.4.6 pre-release
2. Observe MQTT traffic (there won't be any)
3. Downgrade to v2.4.5 (downgrading the firmware is sufficient)
4. Observe MQTT traffic (everything is fine again)
```

rstrouse, 2024-07-14:
Funny, I was performing some testing on v2.4.6-pre and found this issue earlier.  It has to do with the order in which the network connects and the firmware as building with a fix as you submitted the issue.  Re-install v2.4.6-pre and MQTT will connect at the appropriate time.

passibe15, 2024-07-14:
Haha the coincidence. Just installed it and it works as expected.
Thanks so much for the quick fix!