rstrouse/ESPSomfy-RTS #430: missing mqtt sensor attributes

1mfaasj, 2024-07-14:
### Hardware

ESP32

### Firmware version

v2.4.5

### Application version

v2.4.5

### What happened? What did you expect to happen?

Hi,
It looks like that the MQTT sensors in HA are missing attributes. in the normal HA integration (so not MQTT) they are present.
how I would expect it:
[image]

and the mqtt sensor for the same shutter:
[image]
I'm using expecially the following attributes to determines it latest command and timestamp:
```
last_cmd: Up
cmd_fired: 1720983720.445917
```

Is it possible to include the attributes?

### How to reproduce it (step by step)

```bash
I enabled mqtt and it becomes visible in HA.
```

rstrouse, 2024-07-15:
That information is actually sent to the broker but as I understand it the json_attributes for the configuration have been deprecated from the MQTT discovery.  That being said I have no idea why one would want to use MQTT when there is a socket implementation for the device like there is in HA.

1mfaasj, 2024-07-17:
then why do you offer mqtt integration built in? isn't mqtt always used in combination with HA?
in any case, I'll leave mqtt aside for now due to missing attributes. thanks for your response anyway.

rstrouse, 2024-07-17:
There is no need to use MQTT with the HA integration.   There are other automation platforms that do not have the ability to consume the socket that use this as their primary integration though.  If you look in the broker you will find the attributes you desire are actually being sent to the broker but the json_attributes section is no longer supported.  Platforms can subscribe to them as desired.

[image]

1mfaasj, 2024-07-17:
Thanks for the clarification

closed 2024-07-17 by 1mfaasj

rstrouse, 2024-07-17:
Alright I looked deeper into this.  I suspect this has more to do with the state subscription method use in HA.  MQTT is not terribly efficient and as a result the implementation moved this so that the subscription is only one topic.  Sadly this means that if any state attribute changes it requires the entire state payload to be re-published in an effort to improve the efficiency in the HA platform.  

This increases the payload sizes for every device that implements it.  Effectively, it improves efficiency in one place at the sacrifice of another.  In this case it impacts an ESP32 that has very few resources to start with.  The new method uses a json_attributes_topic and a json_attributes_template.   I will take another look at this but unfortunately, it requires dropping the individual mqtt_topics and resending all of the attributes on every change.  This may prove to be too large while it is trying to keep timing between radio frames.