rstrouse/ESPSomfy-RTS #431: Website not responding & device not pingable

geokscott, 2024-07-17:
### Hardware

ESP32

### Firmware version

v2.4.5

### Application version

v2.4.5

### What happened? What did you expect to happen?

Recently upgraded HA to 2.4.7. Router was rebooted today and my device did not reconnect, had to restart. HA integration working fine, but website is no longer responding. Also could not ping device.

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-07-17:
Reboot the device and see if it comes back.  If it does then install v2.4.6 pre-release.  There were changes to accommodate the reconnect.

geokscott, 2024-07-17:
I will wait for the official release. The website finally came back after about 30-60 minutes. Thanks.

ramzan-zz, 2024-07-18:
facing a connection issue after flashing the esp, once visiting the IP address of the ssid espsomfy it shows 'Error opening file'

rstrouse, 2024-07-18:
If you look for wifi networks, does it exist?   If if does not then run the following command.  Give it a few minutes to finish then try again.

`http://<ip address>/updateFirmware?ver=latest`