rstrouse/ESPSomfy-RTS #432: ESP32 (30Pin)+ LAN8710 config

Tpaydt, 2024-07-22:
### Hardware

ESP32

### Firmware version

v.2.4.6 pre

### Application version

v.2.4.6

### What happened? What did you expect to happen?

I would like to connect the LAN8720 board to the ESP32 (30pin version), I have seen that there is a config for this. But so far I had no success, see log. Do I have to modify the LAN8720 like here:
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/
and then activate the PowerPin so that it works with the ESPSomfyRTS?

[image]
[image]

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash
[19:51:40]�������ets Jul 29 2019 12:21:46
[19:51:40]
[19:51:40]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:51:40]configsip: 0, SPIWP:0xee
[19:51:40]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:51:40]mode:DIO, clock div:1
[19:51:40]load:0x3fff0030,len:1344
[19:51:40]load:0x40078000,len:13964
[19:51:40]load:0x40080400,len:3600
[19:51:40]entry 0x400805f0
[19:51:41]E (286) psram: PSRAM ID read error: 0xffffffff
[19:51:41]
[19:51:41]Startup/Boot....
[19:51:41]Mounting File System...
[19:51:41]File system mounted successfully
[19:51:46]Chip Model ESP32-
Preference IP Free Entries: 510
SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
[19:51:46]Connection Type: 2
NTP Settings 
[19:51:46]pool.ntp.org TZ:CEST-1CET,M3.2.0/2:00:00,M11.1.0/2:00:00
[19:51:46]Ethernet Settings
[19:51:46]Board:0 PHYType:0 CLK:2 ADDR:0 PWR:-1 MDC:23 MDIO:18

[19:51:46]Launching web server...
[19:51:46]Creating Web MicroServices...
[19:51:47]Socket Server Started...
[19:51:47]App Version:2.4.6
Starting clean
[19:51:47]Initializing RX Queue
[19:51:47]Connecting to Wired Ethernet
[19:51:47]Set hostname to:
[19:51:47]E (6501) lan87xx: lan87xx_pwrctl(409): power up timeout
[19:51:47]E (6501) lan87xx: lan87xx_init(491): power control failed
[19:51:47]E (6501) esp_eth: esp_eth_driver_install(215): init phy failed
[19:51:47]Ethernet Begin failed
```

rstrouse, 2024-07-22:
Where are you getting those settings from?

From what I see if you wired it like described in the article it should be set like this.

Address: PHY 1
Clock Mode: GPIO0 IN
Power Pin: GPIO 17
MDC Pin: GPIO 23
MDIO Pin: GPIO 18

Tpaydt, 2024-07-23:
the settings are mine, because the ESP32 has no GIPO0 pin, I used GIPO16 instead. Or is that not possible?
I have not yet modified the LAN8710, I thought I would solve the problem with the GIPO16 PIN as CLK mode.
[image]

The main question is do I have to modify the LAN8710 or not?

Or is the ESP32 (30PIN) not suitable for this?

Sorry for the stupid questions, I'm still new to the ESP topic.

rstrouse, 2024-07-23:
The LAN8270 may require a clock in.  I have used the WT32-ETH01 which uses this chip.  

However, GPIO0 on that board is wired up on the boot button and you should be able to run a jumper from there.  That being said you can try to get all of the other settings correct and wire the clock to GPIO16.  The PHY address for the LAN8270 is I2c address 1.

Tpaydt, 2024-07-23:
Thank you very much

i run a jumper and with GPIO0IN it works. 

[image]

rstrouse, 2024-07-24:
Awesome I am going to close this!

closed 2024-07-24 by rstrouse