rstrouse/ESPSomfy-RTS #434: ESP32 WROOM-32D hidden SSID

phatina, 2024-07-24:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

ESPSomfy-RTS firmware won't connect to hidden network. I expected it to work.

### How to reproduce it (step by step)

```bash
1. Disable broadcasting of your SSID.
2. Configure the network settings on ESP32
3. Reboot
```

steigerbalett, 2024-07-26:
It worked till v2.4.5 - maybe you could mention this in the changelog?

rstrouse, 2024-07-26:
I will edit the release notes.  The big difference which started in v2.4.4 is the implementation of the 2.0.17 of Arduino Core.  Prior to this there was no real way to guarantee the strongest connection with multiple access points in a mesh setting.   I had soooooo many reports of the of the AP beamforming to a connection then the ESP32 dropping the connection only to reconnect to a weak access point again.

To do this in Arduino Core you need to identify the AP from the SSID.  In the case where an SSID is hidden the ESP32 cannot identify the strongest AP and will simply connect to whichever AP responds.   This will very often lead to network connection issues for many users.   The footnote on this is that while ESP32 can connect to hidden SSIDs, it really shouldn't until it implements the underlying protocols required to reliably select a beamformed access point.

If however, the motivation behind using a hidden SSID is to improve security you should rethink that.  The SSID is never really hidden and unfortunately, all the devices that need to probe for it make it even more visible (with location data) without ever sending a scan.

rstrouse, 2024-07-27:
I managed to bypass all of the network scanning when connecting to the SSID.  Please install pre-release v2.4.7 and check the checkbox below.

[image]

mgmomo, 2024-07-27:
Hi, perfect, with this it is connecting again to a hidden SSID. Many thx!

rstrouse, 2024-08-01:
I am going to close this one for now.  Let me know if you need anything else.

closed 2024-08-01 by rstrouse

phatina, 2024-08-02:
@rstrouse thank you, works as expected. Good job!