rstrouse/ESPSomfy-RTS #435: Roller Shades only work after using remote

Stooksie, 2024-07-25:
### Hardware

ESP32

### Firmware version

V2.4.5

### Application version

V2.4.5

### What happened? What did you expect to happen?

Expect the shades to move using just ESPSomfy RTS, but I have to use the physical remote first.

### How to reproduce it (step by step)

```bash
1. Try to move a shade using up or down in ESPSomfy.  Nothing happens
2. Use the remote for the shade to move it.
3. Try again using ESPSomfy... shade moves.
4. Wait 5 seconds
5. Try again using ESPsomfy.  Nothing happens

Three shades.  All behaving the same way.

All addresses for the espsomfy remotes are different from each other and from the physical remotes.  All physical remotes linked to the espsomfy remote.

There must be something wrong, but I cannot see what.  The shades are paired and can work, but it seems only immediately after using the remote for the shade. Almost like the remote sends a wakeup command first.

The motors are RTS.

Logs from one of the physical remotes.
```

### Logs

```bash
[{"encKey": 149,
  "address": 12352053,
  "rcode": 218,
  "command": "My+Up+Down",
  "rssi": -76,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 5,
  "pulses": [4008,4001,3995,3518,486,4078,4014,3929,3977,4012,3993,4000,2054,10341,3738,3881,3994,4028,2326,2522,2496,2549,2064,202,351,5278,1256,1335,221,320,493,27,758,1313,1272,1311,1277,1300,444,625,151,202,660,667,602,682,606,1332,603,683,1257,1319,1269,669,622,682,233,336,485,51,102,799,618,1299,632,697,1236,1313,1051,1526,629,668,1183,754,622,669,611,687,611,676,617,653,638,659,589,1349,612,686,615,672,294,489,124,789,626,663,1252,687,101,178,612,671,626,681,610,790,124,174,495,673,614,687,599,1320,610,683,1270,679,157,361,516,775,593,1322,615,740,1207,664,624,1321,103,181,620,663],
  "time": "2024-07-25T16:25:27.796+0100"
},
{"encKey": 174,
  "address": 12352053,
  "rcode": 222,
  "command": "Down",
  "rssi": -68,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [3258,195,342,389,3966,1733,2543,2492,2544,2500,2519,2515,2518,2509,2536,2494,2516,2508,2516,2516,4857,1242,1332,1241,1325,1218,722,607,661,626,1340,1239,679,619,671,619,1308,618,676,614,664,621,671,1275,662,624,692,580,684,609,1328,580,707,607,689,594,677,1281,1316,601,682,1268,663,603,682,615,680,597,675,637,666,624,669,601,1316,614,691,608,716,585,663,619,679,1245,677,620,680,616,668,622,671,608,661,628,672,622,1306,621,664,1290,661,615,666,607,1335,572,722,1235,692,620,1320,609,666],
  "time": "2024-07-25T16:27:18.653+0100"
},
{"encKey": 160,
  "address": 12352053,
  "rcode": 224,
  "command": "Up",
  "rssi": -38,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [3992,2607,76,1375,3978,3989,4007,4081,3995,4014,10475,18005,2540,2497,2520,2509,4857,1270,1313,1270,1305,622,665,630,660,628,664,623,656,1283,1316,614,661,621,668,1284,1316,1246,1314,1271,1308,622,668,631,660,1266,1318,1265,1316,630,661,1277,666,608,1312,1269,1309,1272,1325,1251,668,636,1317,1243,1335,1247,675,614,1310,1272,1313,1270,1310,1272,673,610,1313,643,651,1262,1321,609,687,1275,645,637,1313,618,672,1267],
  "time": "2024-07-25T16:37:35.907+0100"
},
{"encKey": 160,
  "address": 12352053,
  "rcode": 224,
  "command": "Up",
  "rssi": -39,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [638,26930,2511,2524,2513,2524,2511,2524,2518,2496,2542,2486,2519,2515,2521,2505,4852,1258,1336,1246,1316,617,669,636,663,616,665,618,676,1264,1323,609,659,640,650,1272,1314,1270,1309,1272,1295,639,666,628,658,1265,1319,1267,1313,602,682,1268,667,628,1310,1245,1326,1275,1298,1265,667,631,1312,1269,1304,1277,668,610,1311,1271,1312,1267,1306,1273,671,613,1314,617,673,1265,1324,607,661,1265,668,632,1312,618,668,1262],
  "time": "2024-07-25T16:37:35.920+0100"
},
{"encKey": 161,
  "address": 12352053,
  "rcode": 225,
  "command": "Up",
  "rssi": -39,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [2497,2533,2523,2512,4850,1274,1290,1269,1335,596,692,609,661,640,675,1249,660,628,1316,610,689,614,665,1273,1313,1270,659,631,665,618,1298,640,665,618,672,1266,1317,1254,686,620,1294,1284,665,628,1313,1244,1328,1253,1316,1265,667,632,1314,1270,1304,1265,675,614,1316,1270,1313,1269,1307,1274,672,613,1316,618,674,1266,1301,632,661,1279,667,609,1311,619,668,1289],
  "time": "2024-07-25T16:37:39.590+0100"
}]
```

rstrouse, 2024-07-25:
If you are sure you are using unique addresses for ESPSomfy RTS add a couple of repeats to the shade definition as some motors are slow to wake up after they have gone back to sleep.

Stooksie, 2024-07-25:
Thank you.   Perfect fix.  Everything now working

closed 2024-07-25 by Stooksie