rstrouse/ESPSomfy-RTS #437: Add GPIO Pins for SomfyiO Remote controll

Andurilll, 2024-07-26:
### Describe you new feature you'd like

I did my first Setup with a SomfyiO Remote and it works. Yay
I now also soldered the PROG and CH (Channel select) also to wires and would like to controll this from the ESP. Also it would be awesome to read the LED-Status. I also soldered some wires to the LED-input pin. Would love to get infos on the ESP about the selected channel (1, 2, 3, 4, 5=all LED)

rstrouse, 2024-07-26:
I'll have to give this some thought.  To do this the right way, there really would need to be a concept of setting up all the proper GPIO selections.  This is more of a configuration riddle than anything else.   The code to implement it would be very straightforward.

Andurilll, 2024-07-27:
I don't understand what you mean with proper GPIO Selection concept. Is it not possible to use every Pin for everything? (Like now with the dropdown selector)

I guess you could add a "expert Mode" with a Warning and add These there.

If there is a concept even better

Can i help somehow?

(Amazing Work. Thanks. Also the HA-Integration💪🏻)
(What i was wondering:How can these connect to the esp without authentication?)

rstrouse, 2024-07-27:
No I am referring to the UI flow.  Realistically there should be one definition in this case for the remote pins and each channel should be assigned to an individual motor.  If you look at the current structure, the shade definitions all revolve around a unique address.  In this instance there needs to be a page to define the IO-Remote then select the channels from it.

Andurilll, 2024-07-29:
I understand what you mean and agree with you

Andurilll, 2024-08-04:
> No I am referring to the UI flow. Realistically there should be one definition in this case for the remote pins and each channel should be assigned to an individual motor. If you look at the current [...]

What have to admid: there are 1 channel (Situo 1 io) or 5 channel remotes (Situo 5 io) available. Maybe this can be selected somewhere to change available pin options

anno2837293, 2024-09-16:
Hi there,
actually I'm currently working on a way to have temperarure control within my ESPomfy-enhousing to turn off electricity in case ot gets way too warm. First idea was to set up a separate ESP8266 just for that, which seems to be quite an overkill. As I'd need only one readout pin I thought I'd join this request and ask for some kind of expert mode where the use of a few of the free pins could be implemented.

I'm planning on using a DHT22 so besides a 3V3 and GND there would only be one readout pin that would be used.

As already mentioned in previous comments: AMAZING WORK and thanks so much for it! :)

All the best,
Anton