rstrouse/ESPSomfy-RTS #438: Version 2.4.6 won't connect to Hidden Access Points

edrose, 2024-07-27:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

After updating to firmware 2.4.6 I started having issues getting the ESP32 to connect to my IoT network, which is a hidden WiFi network with some of the more modern WiFi features turned off so it plays more nicely with IoT devices. It seems like the logic for reconnecting to WiFi has changed, and if the SSID doesn't appear in a scan list then it never attempts to reconnect.

When you add a hidden network to an Android phone there is a checkbox to mark the network as hidden, I think to solve this exact problem. It forces a reconnect attempt, regardless of whether the network appears in a scan. Do you think a checkbox like this would be possible to implement?

### How to reproduce it (step by step)

```bash
1. Setup v2.4.5 on a WiFi network with a hidden SSID
2. Update the firmware to v2.4.6
3. The SoftAP will be started and the device will not connect
```

m4um4u1, 2024-07-27:
Just want to approve the issue. Im also using a hidden IoT network

edrose, 2024-07-27:
> There are many more networks out there with mesh and beamforming as reflected by the periodic disconnects for other users.

Ironically my hidden IoT network was created was to improve connection stability by disabling all of the fancy modern stuff and leaving just a bog-standard WIFi network! It's been rock-solid until now.

>  If the ability to ensure the strongest AP on the SSID is hampered then I am afraid that hidden SSIDs won't be supported.

Roaming between APs almost never works on hidden SSIDs regardless of the device, but if there is only a single AP then roaming is irrelevant. It's just a different connection strategy and it's a shame that it's not all abstracted away by the ESP32 libraries.

---

Can I suggest adding a comment to the release notes for anyone else updating, to warn them about this change in behaviour. It was not obvious what the issue was on my end, and I spent quite a bit of time re-flashing the update and banging my head against the wall about why it would no-longer connect to my WiFi.

rstrouse, 2024-07-27:
I am currently looking at bypassing the autoReconnect and SSID optimization using a checkbox on the network tab.  The network connectivity stuff for the ESP32 has been a bit of a bugger.  I believe most of it stems from the lack of protocol support for mesh networks which probably didn't exist at any appreciable level when the radio was developed.  I believe there are probably some hardware restrictions with it.

> Can I suggest adding a comment to the release notes for anyone else updating, to warn them about this change in behaviour. It was not obvious what the issue was on my end, and I spent quite a bit of [...]

It was never my intent to make hidden SSIDs not work.  The intent was to make sure that the ESP32 connected to the strongest AP when it was available.   The issue with many was that when their network rebooted it would connect to the first AP that came up.  Unfortunately, this was often the master router.   If they had additional APs then the ESP32 would end up bound to the wrong AP.

I will add a note to the release notes indicating that in v2.4.4+, hidden networks can be an issue.

rstrouse, 2024-07-27:
I managed to bypass all of the network scanning when connecting to the SSID.  Please install pre-release v2.4.7 and check the checkbox below.

[image]

edrose, 2024-07-28:
Just installed 2.4.7 pre-release, and that option seems to be working for me. Thank you for implementing a fix so quickly!

rstrouse, 2024-08-01:
I am going to close this one for now let me know if you need anything else.

closed 2024-08-01 by rstrouse