rstrouse/ESPSomfy-RTS #439: my / favorite position Blinds

moshiko2312, 2024-07-29:
### Hardware

ESP32, Other

### Firmware version

v.2.4.6

### Application version

v2.4.7

### What happened? What did you expect to happen?

I have 10 shaders in the project with a tilt position. I'm trying to access my button, for example shader at 0% and the tilt at 10%. What actually happens is the shader drops to 0% but does not move the tilt to 10%, it only works if I close the shader manually to 0% and then I press my button so the shadow performs the 10% tilt but from any other position it doesn't happen.
Home Assistant also shows other values ​​in percentages than what appears in the application.

I'm trying to figure out what I'm doing wrong
I set up and down times for each shade separately

There is also an issue that if I press to move only the tilt from Home Assistant sometimes it just takes the tilt to 100% instead of going step by step with each press

[image]
[image]

[image]

### How to reproduce it (step by step)

```bash
1. I keep a position for each shade separately/

2. I press my button, the operation is carried out, but the shade does not open to the 10% that was set for it.

3. If I manually lower the shade to 0% and then press my button, it does work.

4. I think there is no correlation between the percentages and the motor itself.
Although keeping the position in my sewing machine is burned in the motor itself.
I can't figure out why it really doesn't work.
```

rstrouse, 2024-07-29:
It sounds like you have a couple of things going on.  First, does the My button work appropriately from the remote?  If it does I would like to see the frames being sent from the radio log.  If it does not, do you know the brand and model of the motors?  Some of these motors do strange things with the My operation but you can try checking the simulate favorite option and ESPSomfy RTS will issue radio commands in sequence to affect the position.

The other issue sounds a lot like the motor is set to tilt priority or euro mode.  In this mode a short press of the up and down only moves the motor a short distance.  It sounds similar to this issue #395 is this what you are experiencing?

moshiko2312, 2024-07-30:
I will try to check again the properties.
Because this restaurant is 3 hours from my office so it's not easy.

Can you please write for me list what exactly i need to check and im going there.
Thanks.
Or if have some way to contact you directly like WhatsApp or Facebook.
This is my email if you need write to me.
moshiko2312@gmail.com
Thanks for your all job.
🙏

moshiko2312, 2024-08-05:
[image]
[image]

rstrouse, 2024-08-05:
First set each of the motors to 80-bit RTS with a tilt type of Integrated.  Then, set the step size to 100.  This is simply a ratio of the overall tilt time that is each step affects the tilt.  Once you get everything else figured out then you can tweak these settings.

After you have done that, open the virtual remote and test the up and down buttons for each blind.  If pressing the up or down button only moves the motor a little bit.  Move the blind to 50% with the remote then press and hold the UP + DOWN in the virtual remote until the blind jogs.  Test the up/down buttons again.  In the end, the up and down buttons in ESPSomfy RTS should move the motor between 0% and 100%.