rstrouse/ESPSomfy-RTS #442: Is it me, or Somfy Exavia can not be paired?

mpietruszko, 2024-08-01:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

I'm having difficutlies in pairing the ESP-WROOM-32 + E07-M1101D-SMA with Somfy Exavia. Tried multiple shade types in pairing modes and by using the virtual keyboard to reproduce the toggle button, both in 56 and 80 bits. 

The pairing process for the central is to wake up the device and click the button on the remote once (https://youtu.be/EK_iJpB9nmI?si=tOvrYVxiS2TPDpy0&t=272), which works with the remotes that I have. The ESPSomfy logs all performed actions as the Toggle command. 

[image]

Any suggestions or ideas how to move forward? Thanks in advance. 

### How to reproduce it (step by step)

```bash
1. Wake up the Exavia central
2. Click pair/toggle in ESPSomfy
3. No feedback received from the device
```

rstrouse, 2024-08-01:
Try using the Virtual remote after setting the Exavia into pairing mode then press and hold the toggle button.   I cannot read the manual for it, but it does indicate it is capable of using the 3 button remote.  However, it also indicates that it will only pair when the radio is very close to the unit.   Move ESPSomfy RTS close to the pairing location as indicated in the manual.

This should work just fine.

rstrouse, 2024-08-01:
I take that back, from the pictures it looks like it pairs by holding down the UP button.

mpietruszko, 2024-08-03:
Thanks for a quick reply, it stopped raining so had a round two with the gate. Sadly, no progress at all. 

Tried holding/clicking both the Toggle button (since the gate comes with two buttons remote which identifies as toggle) and the recommended UP button for the 3 button remote. No matter if the antena was close or far, the central did not respond for the virtual remote buttons and always did for clicking the remote close. 

Not sure if it has something to do with the power, but I assume that the ESP dongle is way more powerful than the PCB antena from the remote. Tried to keep it close, tried to keep it far, but no LED respond from the gate once in the pairing mode. Successfully paired the remote, but it has probably nothing to do with the pairing functionality. Triple checked the TX wiring and it looks good. No idea what goes wrong, hard to find any extra information about this particular device.

rstrouse, 2024-08-03:
This should work if there are pairing slots available.  While using the Virtual Remote you may need to hold down the button.  Some of these devices use a weird medium length press for the pairing.  

Also, I know you said you triple checked the wiring but I have experienced several instances where folks have felt their wiring was correct only to have them look again and realize the RX and TX lines were not correct.   It is important that PIN3 on the transceiver is connected to a GPIO and that GPIO is set on the TX dropdown in the configuration.   If it is wrong it will receive but not transmit.

Another thing that I have seen are instances where an address has been registered with the motor but the rolling code in the motor is not correct so it cannot be used.  Just to be sure, you did not set the ESPSomfy RTS address for the gate to one that was used in the past with another integration or by an existing remote.

mpietruszko, 2024-08-05:
Okay, what can I say. When a man says to triple check, quadruple check - I fully focused on the diagram from the Wiki and the ESP itself, but never checked what ESPSomfy had in the radio settings and took it for granted.. At the end GPIO 12 and 13 needed to switch places in the settings. 

Thanks again for delivering such an awesome solution to solve many people daily challenges and spending your time to answer silly posts like mine.

closed 2024-08-07 by mpietruszko