rstrouse/ESPSomfy-RTS #443: I broke the pairing of my Telis 4 remote control and need help fixing this issue and configuring ESPSomfy

michaelmell, 2024-08-01:
Dear all,

first of all I want to say thank you for this great project. I was able to install ESPSomfy RTS on an ESP32 WROOM, set everything up, and start controlling my two awnings without issue. The documentation is very detailed and helpful.

Still I was somehow able to mess things up and I now need help to resolve the situation I am in. I will start out by describing (1) the configuration that I would like to achieve, (2) what I did, and (3) the issue I am now facing...

## 1. What I want to achieve:

I have two awnings that I am controlling with a 5-channel Telis 4 remote. I would like to use both the Telis 4 and ESPSomfy to control the awning. ESPSomfy should pick up any commands sent by the Telis in order to track the state of the awnings.

## 2. Here is what I did during setup:

In its original configuration (before I started messing things up), the Telis was setup like this:

channel 1: left awning
channel 2: right awning
channel 3: both awnings (this was used almost exclusively)

To setup up ESPSomfy, I first read the remote address from the radio log for channel 3 (i.e. for both awnings). I setup an awning device using this radio address /without/ doing any additional pairing (i.e. I did not press the "Pair Shade" button). 

After this I could initially control the awnings both with the remote and ESPSomfy. But I saw weird behavior in that the awning sometimes did not respond to commands from either ESPSomfy or the Telis.

## 3. The issue I am now facing:
After a while channel 3 of the Telis stopped working completely, while I am still able to control the awning with ESPSomfy. I suspect that this is due to conflicts from the rolling-codes.

I have since been able to use the "Virtual Remote" in ESPSomfy to put both awnings into pairing mode and pair them again with channel 5 of the Telis.

But I am not able to pair them with channel 3 again. For some reason the awnings do not jog, when tying to pair channel 3 again (as they did, when I performed the same steps as I did for paring channel 5). I suspect this is because channel 3 is still paired with the awnings (even though they do not respond to it). Trying to un-pair them also does not work. I follow this Youtube-video for the pairing- and un-pairing-steps: https://www.youtube.com/watch?v=gZaWhwjbK6g

Any help fixing this situation and advice for achieving the desired configuration described above would be greatly appreciated.

Thank you in advance and kind regards,
Michael

rstrouse, 2024-08-01:
>  To setup up ESPSomfy, I first read the remote address from the radio log for channel 3 (i.e. for both awnings). I setup an awning device using this radio address /without/ doing any additional pairing (i.e. I did not press the "Pair Shade" button).

This will ALWAYS break either ESPSomfy RTS or the Remote.   In many cases you will need to factory reset the motor then use unique addresses for ESPSomfy RTS and your remotes.  Unfortunately, when the rolling code for the address gets out of sync with the remote it will no longer pair/unpair or control the motor.

Please factory reset your awning motors and set the endpoint limits for them.  If you look online for the documentation on your awning you will find this procedure.  Next pair up all your remote channels as you would like them.    After you have done that delete all the shade definitions from ESPSomfy RTS.

When you add the shade definitions back in use the pairing function and do not change the address to the address of an existing remote.   EPSomfy RTS does its best to ensure the addresses are unique.  I know there are a bunch of projects out there that do this but the expectation is that you can no longer use your existing remotes.

After you have everything paired up go back to ESPSomfy RTS and link your handheld remotes to the shade definition using the link remote button.  This will ensure ESPSomfy RTS hears every movement and can keep track of the motor position.

Think of ESPSomfy RTS as a 48 channel remote that has 32 single channels and 16 groupable ones.   Each address that communicates with the motor must be unique and this includes your remotes.  The key difference between a dumb remote is that ESPSomfy RTS has a receiver that can hear those remotes as well.

michaelmell, 2024-08-01:
Thanks so much for your explanation. I did not understand this from the ESPSomfy documentation. But, admittedly, I did only read the ESPSomfy documentation and did not check any additional instructions on how to do pairing / programming. Given that it uses rolling codes and had I given it a bit more thought, then I would have realized that this would break .

I am bit anxious on doing a factory reset of the awnings, because they were pre-installed in a flat that we bought and I do not have any documentation on them. Furthermore, they are paired with a wind-sensor that I will also have to setup.

I did see these two Youtube videos that explain nicely how to do the factory reset (by doing a power-cycle with 3-5-3 seconds) and programming (so I may give it a go):
reset: https://www.youtube.com/watch?v=sTGraGEfEH4
programming: https://www.youtube.com/watch?v=z3FQa_RRXKw

Regarding the wind-sensor:
I have this wind-sensor (manual in German): https://service.somfy.com/downloads/at_v5/solirissensor_rts_led_ga_deutsch.pdf.pdf

From briefly scanning the manual, I see that it is a wind and sun sensor. I saw in ESPSomfy that shade can be configured with wind and sun-sensor. Given that I was already able to mess this up once...
Could you briefly explain to me I would set this sensor in ESPSomfy RTS? My current understanding is that you only need to select the two "checkboxes" for the wind- and sun-sensor...

Thank you again for the helpful and quick reply,
Michael

rstrouse, 2024-08-01:
Sensor linking is very easy and takes less than a minute to set it up.  By defining the motor as an awning it will listen for all wind input.  Simply check the Has Sun Sensor checkbox and it will allow you to enable and disable the sun sensor for the awning.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#linking-sensors

To link the Soliris is remarkably simple as well with a sequence of button pressing for the motor and the Soliris.