rstrouse/ESPSomfy-RTS #444: Network connection interrupts

HamsenBender, 2024-08-02:
### Hardware

ESP32, LilyGO T-Internet POE

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

I see in HA that the device reconnects and updates all sensors every few minutes, it reports them unavailable first.
This happens randomly, but also when sending a command to open or close one of the shades.
When I ping my device, I also see a drop from time to time. At first I thought the WiFi was unstable, even though I have plenty of other ESP devices which are stable. But after replacing the device with a wired one (LiliGO) the issue still occurs.

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash
2024-08-02 13:33:02.155 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [281472386643232] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2731, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2774, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/cover.py", line 788, in async_set_cover_position
    await self._controller.api.position_shade(
  File "/config/custom_components/espsomfy_rts/controller.py", line 808, in position_shade
    await self.shade_command({"shadeId": shade_id, "target": position})
  File "/config/custom_components/espsomfy_rts/controller.py", line 818, in shade_command
    await self.put_command(API_SHADECOMMAND, data)
  File "/config/custom_components/espsomfy_rts/controller.py", line 842, in put_command
    async with self._session.put(f"{self._api_url}{command}", json=data) as resp:
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client.py", line 1197, in __aenter__
    self._resp = await self._coro
                 ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client.py", line 608, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client_reqrep.py", line 971, in start
    with self._timer:
  File "/usr/local/lib/python3.12/site-packages/aiohttp/helpers.py", line 735, in __exit__
    raise asyncio.TimeoutError from None
TimeoutError

2024-08-02 14:02:29.155 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [281472388502944] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2731, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2774, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/cover.py", line 806, in async_open_cover
    await self._controller.api.open_shade(self._shade_id)
  File "/config/custom_components/espsomfy_rts/controller.py", line 779, in open_shade
    await self.shade_command({"shadeId": shade_id, "command": "up"})
  File "/config/custom_components/espsomfy_rts/controller.py", line 818, in shade_command
    await self.put_command(API_SHADECOMMAND, data)
  File "/config/custom_components/espsomfy_rts/controller.py", line 842, in put_command
    async with self._session.put(f"{self._api_url}{command}", json=data) as resp:
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client.py", line 1197, in __aenter__
    self._resp = await self._coro
                 ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client.py", line 608, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client_reqrep.py", line 971, in start
    with self._timer:
  File "/usr/local/lib/python3.12/site-packages/aiohttp/helpers.py", line 735, in __exit__
    raise asyncio.TimeoutError from None
TimeoutError
```

rstrouse, 2024-08-02:
This looks like it is timing out sending a request.  Tell me a little bit about your Home Assistant installation.

rstrouse, 2024-08-04:
Also, are you using the IP address or the hostname.  Unfortunately, some routers do not do a very good job with mDNS as it is implemented on the ESP32.  This is especially so when there is not an IP reservation in your router.  I do not experience any of these issues so it has to be something different in your environment.

closed 2024-08-08 by HamsenBender