rstrouse/ESPSomfy-RTS #445: Wifi fallback does not work at first boot

Jezza34000, 2024-08-04:
### Hardware

ESP32 WT32-ETH01

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Hi Robert,
My board is configured :
-Use Ethernet
-Fallback to Wireless
Working OK with ETH, Working OK with Wifi
ETH -> Wifi : Fallback is working OK
But if you start the board with ETH disconnected, it will never connect to Wifi

### How to reproduce it (step by step)

```bash
1- Start board with ETH disconnected -> it never fallback to wifi
```

rstrouse, 2024-08-04:
Interesting.  Did you wait at least 30 seconds for the wired delay to kick in.  In this initial condition users have asked that there be a delay prior to attempting to start the wireless connection to allow for managed switches and the like to come online during a power loss.

Either way I will set up this scenario and try it out.

rstrouse, 2024-08-04:
I tried this out and it is waiting for the adapter to respond.  I will look at modifying the timeout to fallback when there was not a wired connection to begin with.

rstrouse, 2024-08-04:
v2.4.7 - Pre works around not getting a disconnected output from the wired connection if it has never been established.  Interestingly this is not the case for a wireless device.  It always reports a disconnect if the connection failed.  Either way if nothing can be established it will stop the connection attempt after 20 seconds.  Install the v2.4.7 pre-release and let me know if it fixes your issue.

Jezza34000, 2024-08-04:
Your are so fast !
i'll install this and i try....

Jezza34000, 2024-08-04:
Tested & OK : Fixed into 2.4.7 👍🏻

closed 2024-11-17 by Jezza34000