rstrouse/ESPSomfy-RTS #446: poor wifi signal

leonardpitzu, 2024-08-04:
### Hardware

ESP32, ESP32-C3

### Firmware version

2.4.5

### Application version

2.4.5

### What happened? What did you expect to happen?

Not sure if it's a bug but wifi signal is -91dBm (reported by unify controller) although the board is 20cm away from the AP. This has gotten worse but no idea why. Any tips on debugging? I'dd try a second board but I have no radio modules and it takes about a month until I receive the new order. The board is not "made in china" so I guess the antenna "should be ok" (can't swear to it though...).
Have a bunch of esp32's spread throughout the house but none are the same so bad luck in comparing them. Network is stable though as no other device reports such poor signal.

### How to reproduce it (step by step)

```bash
1. flash
2. configure
3. crappy wifi signal
```

### Logs

```bash
n.a.
```

rstrouse, 2024-08-05:
Which c3 module do you have?  Does it have an external antenna?

leonardpitzu, 2024-08-05:
No external antenna (not even a connector) - just ca ceramic one.

rstrouse, 2024-08-05:
Do you have the ceramic antenna covered by anything?  Do you have more than one AP and are you sure it is connecting to the close one?  Are you seeing -91dBm on ESPSomfy RTS config pages as well?

Ceramic antennas are also very susceptible to interference.   However, if there is nothing covering the antenna chip I would point to a defective part.

leonardpitzu, 2024-08-05:
The antenna is not covered in anything and is close to the AP. Yes, there are more AP’s but the SoC is connected to the closest one (as reported by unifi controller). Unifi controller reports -91dBm and ESPSomfy -50dBm but the . I would “trust” the controller as its reported value is consistent with the dropouts I am experiencing.

rstrouse, 2024-08-05:
That is odd that there would be such a discrepancy between the LQI over the two.  Please update to v2.4.6 and turn off roaming if you are specifying an AP through binding.  If you use a hidden AP use v2.4.7 pre-release and turn on Use Hidden AP.

By chance you your have 802.11b enabled on that network?

passibe15, 2024-08-07:
Interesting that someone else also experiences this – I thought it was just a peculiarity of my AP (Zyxel NWA210AX).
But I've also had this issue with every firmware version I have used so far (the first version I used was v2.2.2, i think).
For example right now, the ESP's web interface reports -48dBm while the AP reports -90dBm. Interestingly, right after a reboot of the ESP the signal strength reported by the AP is slightly better (around -85dBm, but I remember it sometimes even being much higher and very close to what the ESP was reporting) and then within a few minutes it drops to around -90dBm, where it stabilizes.

The ESP and the AP are very close together, so I believe the -48dBm is the "correct" one. That's also the reason I thought it was just this particular AP reporting it wrongly, because -48dBm seems very plausible given the ESP's proximity to the AP.

I am using v2.4.6 with roaming turned off, 802.11b is not enabled. Also, I am not experiencing any dropouts.

If you need any other info let me know!

leonardpitzu, 2024-08-07:
802.11b is disabled. I run the latest "beta" firmware and turned roaming off. there is still a discrepancy between the ESP reported RSSI and the one from the unified controller. I made a very short test and flashed esphome on the board and the unified reported rssi is the same while the esphome reported one is closer (but still not identical). In any case the signal sucks major. I now have a hunch that this is due to the close proximity to the AP which causes the antenna to be flooded in "noise". I need some tome to test this out as the board and the AP are in the ceiling...

rstrouse, 2024-08-11:
If the two radios are very close then you may be experiencing saturation.  This will bleed the channels together and destroy the connectivity with near field distortion.  As for the RSSI this is a calculation of LQI and will be different between the transmitter and receiver in many cases.

passibe15, 2024-08-11:
Yes, sure, a slight difference between the transmitter and receiver is expected. But a difference of 42 dBm seems a bit much.

Interestingly, for other devices this is not an issue (I have a HomePod mini and two ESP8266-based devices also very close to the AP). For those, the AP reports between -36 (the ESP8266s) and -30 dBm (HomePod).

But maybe this is something specific due to the ESP32's board design?

I don't have another one lying around to test at the moment, but I'll try moving the Somfy one further away and see if the values make more sense then.

rstrouse, 2024-08-11:
> Yes, sure, a slight difference between the transmitter and receiver is expected. But a difference of 42 dBm seems a bit much.

Actually the LQI is only measured on the receiver so this is an indication that while the ESP32 is hearing the the AP it is resulting in resends when responding to the AP.  That being said the interweb has quite a few articles discussing ESP32 wifi disconnect problems.  Many of these seem to boil down to either power on the board or incompatibilities with the AP options like BSS and 802.11b packets.

passibe15, 2024-08-11:
> so this is an indication that while the ESP32 is hearing the the AP it is resulting in resends when responding to the AP.

Ok, thanks! That makes sense. I'll read up on this.

leonardpitzu, 2024-08-20:
So after a lot of fiddling around and replacing the ESP32 board with another, identical one, I figured I might as well try another board/SoC. I switched to the Speed Studio Xiao ESP32S3 with an external WiFi antenna. The firmware and overall setup (including distance to the AP) remaind exactly the same. I only replaced the ESP32 in the case where the radio was already in. The difference is immense - the ESP is able to cope with the noise and channel way better and the reported values by the Unifi controller and ESPsomfy-RTS are almost identical.
From my point of view the issue can be closed as it is definitely a HW issue in the sense that ceramic antennas simply are not enough.
Price was low enough (so low I even bought 10 board to replace all the other ESP32's I have around my house) so I can only suggest you all do the same and at least have a proper HW setup.