rstrouse/ESPSomfy-RTS #447: Frequent Disconnects

JeffSteinbok, 2024-08-09:
### Hardware

ESP32

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

My device is continually dropping off the network and coming back:
[image]

I've turned off roaming and saved, still doing it.
Only time I've seen something like this on this particular network was one appliance (a washer) which was getting too much traffic and causing it to disconnect. 

Is there any way to diagnose what may be going on here?

Also, oddly in Unifi, I don't see an IP address for it, even though it does have one.

### How to reproduce it (step by step)

```bash
1. Turn on
2. Wait
```

### Logs

```bash
Where can I find logs?
```

rstrouse, 2024-08-09:
Turn off ssdp discovery and mqtt if you have them on.

JeffSteinbok, 2024-08-09:
MQTT is off. 
Publish Discovery is off. 

Are there ways to get logs?

rstrouse, 2024-08-09:
If you connect it to usb you can see the logs in the esphome web site.

JeffSteinbok, 2024-08-09:
I went ahead and put the device on a smaller VLAN where I have less broadcast traffic. So far, more stable. 

I would like to see if this is something in the firmware or ESP chip rather than network.

JeffSteinbok, 2024-08-09:
> If you connect it to usb you can see the logs in the esphome web site.

Do the logs survive reboots? Or is it just looking at the log output live?

JeffSteinbok, 2024-08-09:
Looks like this maybe?
I have two devices that seem to be impacted. 

https://github.com/Aircoookie/WLED/issues/3398

rstrouse, 2024-08-09:
The logs are live and do not survive a reboot.  What do you have that may be sending large numbers of UDP packets around on your network?

naftalih, 2024-08-11:
Having same problem. ESP32 Wroom board + firmware 2.4.6

This project is great but unusable when it restarts all the time....

rstrouse, 2024-08-11:
@JeffSteinbok 
> Also, oddly in Unifi, I don't see an IP address for it, even though it does have one.

Have you assigned this as a static IP?  Could you by chance have an IP conflict?  Typically your best bet is to create IP reservations and let DHCP handle it.  Also, is your router channel hopping.  The ESP32 is not a 2x2 and will need to disconnect/reconnect on a channel change.

@naftalih - I have no less than 10 ESPSomfy RTS devices connected which include all of the supported chip series and they never disconnect with the latest employed Arduino libraries used in ESPSomfy RTS.  Earlier libraries had issues connecting to Multi-AP systems in that it would connect to the first responder whether that was the best AP or not.

Start by disabling UPnP over SSDP, MQTT, and roaming if you have them enabled just to make sure there is not some other communication issue here.  Another user was running some Bluetooth sensors that also interfered with their WiFi connection.  

> Update: I have changed the wifi connection from 2.4 network to 5 network. Will report back...

Unfortunately, you are limited to the 2.4Ghz band for ESP32 wifi for now.

phillipbkemp, 2024-12-24:
Disabling 'Broadcast uPnP over SSDP' seems to have sorted this problem out for me.  Thanks all!

agekristian, 2026-01-16:
Hi
When I turn off "Auto Check for Updates" it stops restarting?

cooLopke, 2026-01-17:
I've got the same problem. Every 1 minute the device disconnects. I also tried to connect it to my phone hotspot. Then it seems a little bit more stable, but still a lot of requests timeout. Great project, but at this rate sadly unusual. Is there any way to disable power save mode? I've read a lot about doing it in esphome and it should fix the problem.

I'm using an ESP32D