rstrouse/ESPSomfy-RTS #449: Try to reconect if wifi is not connected

lukeskymuh, 2024-08-12:
### Describe you new feature you'd like

Hi,

This project  is amazing! I just have one issue: when the power is off and returns the wifi router requires some time to boot.  Aparently the ESP32 is faster and doesn't find the Network, It doesn't try to connect again and stays unconnected. Is there a way to automatically to try again to connect if there is no connection?

It is powered via USB to a resberry pi running home assistant.

BR Luke

rstrouse, 2024-08-12:
Which firmware are you using?  It should connect as soon as it sees the AP in the scan list.  It will do this continuously unless you have it set to autoconnect to the AP SSID.

lukeskymuh, 2024-08-16:
V 2.4.4
Where can I find the autoconnect option?

rstrouse, 2024-08-16:
Pleas update your firmware.   The connection options to reconnect to the strongest signal did not exist in v2.4.4 on bootup.

Nickduino, 2024-11-12:
> I have a similar problem:
> once my router reboots in the mesh, the esp does not reconnect (I have to power cycle it).

I thought it had been fixed but I'm experiencing similar behavior with v2.4.6. In the last month, I've had to reboot my router twice and both times the ESPSomfy reverted back to softAP() mode and never reconnected to the WiFi network (which is **not** hidden and is 2.4 GHz only) by itself. The first time I power cycled the ESP32; the second time, I connected to the AP and asked it to reboot (BTW, have you considered making the page a [captive portal](https://medium.com/@atacanymc/creating-a-captive-portal-with-esp32-a-step-by-step-guide-9e9f78ab87b8), so that we don't need to remember http://192.168.4.1/?)

> I am on 2.4.6 and I am unable to find any auto reconnect option.
> @rstrouse can you guide me to this option?

It's not an option, is should be the default behavior.

Nickduino, 2024-12-03:
> In the last month, I've had to reboot my router twice and both times the ESPSomfy reverted back to softAP() mode and never reconnected to the WiFi network (which is **not** hidden and is 2.4 GHz [...]

@rstrouse Same today, I rebooted the router 90 min ago and ESPSomfy RTS has not reported for duty