rstrouse/ESPSomfy-RTS #450: Tuple Index out of range?

Pleeriezer, 2024-08-13:
### Hardware

ESP32

### Firmware version

2.4.3

### Application version

2.4.2

### What happened? What did you expect to happen?

Hello,

I get a message in HA that there is a new version of he software.
Tried to install it but I am getting te message Tuple Index out of range.
And I can not install it?

How can I install the latest version?

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash
No log available
```

rstrouse, 2024-08-13:
Unfortunately, HA did an update that made breaking changes that have been fixed in v2.4.7 of the HA plugin.  If you are using HACS simply go to the HACS store in HA and select ESPSomfy RTS.  On the upper right you will see a 3 dot menu.  Click on redownload and install v2.4.7,

Pleeriezer, 2024-08-13:
Thanks for the quick reply.

Unfortunately I see no espsomfy in my HACS?

[image]

rstrouse, 2024-08-13:
Did you install it from HACS or did you copy the files to the custom_components directory manually?

Pleeriezer, 2024-08-14:
I installed it a few years ago trough HACS?

rstrouse, 2024-08-14:
Hmmm.... when you select the filter to Downloaded on the right and clear the search text does HACS show up?  I am running 1.34.  Also do you have an espsomfy_rts directory in the homeassistant/custom_components/ directory?

Pleeriezer, 2024-08-15:
Yes I have :

[image]

[image]

[image]

Pleeriezer, 2024-08-16:
Hello,

I have updated it trough Github, so I hope that in the future it will be good.
Thnx for the help!

closed 2024-08-16 by Pleeriezer