rstrouse/ESPSomfy-RTS #452: I mess up pairing process and I need advice

froxic, 2024-08-14:
Hi, firstly thank you for nice project. Tutorial for HW assembly was really nice and it was quite easy to do initial setup. Thing is that I was reading manual in some time later and I was thinking that Im doing it according the instructions. Well, I was wrong and messed step during adding blinds. 

What I did - I entered same address for each blind as Somfy RTS - I know that this was stupid decision. But at the current state I lost 3 channels of 5 on Telis remote, two of them Im able to operate from ESP, one is not working either from ESP or Telis. Rest two are fine, at lest those for controlling one floor and whole house, so I did not lost Telis remote function completely. I was thinking now to do the same in reverse, so I would like to program blind from ESP to Telis, It shakes for first two channels but Im not able to restore Telis remote functionality.

Another issue is that all blinds on house - 6 in total are in the same electric circuit so Im not able to perform reset process in easy way. Blinds are also in a flush installation in cassette under facade so its not easy to get to the motor but I assume it will be doable with some effort, installation company did replacement of one blind string few years ago.

Any comments, thoughts?
Thank you in advance.

rstrouse, 2024-08-14:
The motors that are not responding to the remote will not unpair or pair on that channel until the motor rolling code matches the rolling code being sent from the remote.  In some cases repeatedly pressing the remote can sometimes bring back the channel.  However, eventually it will stop responding to that address until you factory reset the motor.

When developing this solution I had a similar problem except mine are very heavy and high in the air.  Initially, I didn't understand the rolling codes and used the same addresses.  

My rollers are 4 inch stainless steel and they are all wired together.  I ended up having to reset them by connecting and disconnecting them at the junction box for each motor.  Most often you will find that the motor wires connect to the circuit in a short run from the motor using wire nuts.   In the US these must be in an accessible junction box.

froxic, 2024-08-15:
Thanks to your advise (rapid button mashing) I was able to get back 2 channels on Talis remote, so immediately I deleted old config from ESP Somfy and added new virtual remotes with proper addresses. Now its working from both. Problem is now only with 3rd motor / channel. For this one I think I need to do proper reset...

rstrouse, 2024-08-16:
Yeah it does sound like a factory reset on the 3rd motor is required.

leonardpitzu, 2024-08-20:
I have one motor with a missing slot. Somehow it works with 3 out of 4 remotes. Is there any way to reset the motor using the working remotes or do I have to reset it with the power toggle and repair the remotes all over again?

sk-ilya, 2024-08-28:
Yeah, I've been there. The project is awesome, but it's just me being stupid—I didn't read the manual properly and thought linking the ESP to the original remote's channel made sense. I had to spend a couple of hours connecting and disconnecting the motors that were joined at a single junction box.

@rstrouse, things went way smoother on my second attempt to connect the motors after I watched a couple of YouTube tutorials about programming the remote and played with the original one, trying to switch channels for the motors. After that, I became way more confident and it started to be very clear what was going on during the linking process. Maybe it makes sense to add a recommendation to watch a video tutorial on the programming process to the "linking motors" section?

hvltaj, 2025-07-10:
> The motors that are not responding to the remote will not unpair or pair on that channel until the motor rolling code matches the rolling code being sent from the remote. In some cases repeatedly [...]

I have a similar situation, multiple blinds on same circuit, with multiple circuits. And few of those blinds stopped responding to the control long time age. I wasted huge amount of time trying to figure out how to reset the blind one by one without taking cover boxes out and cutting the power near the motor.

And then I learned from you that they use rolling codes, and trying many times may fix them. 

It seems to be working, at least on two of them which I've tried!

Many thanks, you've made my day!

PS. for other searching how to fix the same issue, I've tried clicking the pilot multiple times after reseting power circuit breaker. This also could be a factor?, because I'm pretty sure I tried spamming the pilot in the past.