rstrouse/ESPSomfy-RTS #453: Not connecting to WiFi

DrmedAllel, 2024-08-14:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Flashed Firmware to the ESP and done all the setup via the ESPs Wifi. After the reboot the website is not available via my local Wifi. Looked up in the monitor and realized that there is a loop while the esp tries to connect to my wifi.

### How to reproduce it (step by step)

```bash
1. Flash ESP
2. Enter Wifi SSID and password
3. reboot
```

### Logs

```bash
16:45:29.865 -> Connection Type: 1
16:45:29.865 -> NTP Settings 
16:45:29.865 -> pool.ntp.org TZ:
16:45:29.865 -> WIFI Settings
16:45:29.865 ->  SSID: [***********] PassPhrase: [**********]
16:45:29.865 -> 
16:45:29.865 -> Launching web server...
16:45:29.865 -> Creating Web MicroServices...
16:45:30.510 -> WiFi Mode: 0
16:45:30.510 -> (evt) WiFi interface ready
16:45:31.018 -> WiFi station mode started
16:45:31.018 -> Socket Server Started...
16:45:31.018 -> App Version:2.4.6
16:45:31.018 -> Starting clean
16:45:31.018 -> Initializing RX Queue
16:45:34.820 -> (evt) Completed scan for access points (3)
16:45:34.820 -> Connecting to AP 38:10:D5:1A:64:0A CH: 11
16:45:35.365 -> WiFi begin...
16:45:35.365 -> Timing Net: 110ms
16:45:53.487 -> (evt) Disconnected from WiFi STA access point. Connecting: 1
16:46:11.900 -> (evt) Disconnected from WiFi STA access point. Connecting: 0
16:46:16.136 -> (evt) Completed scan for access points (3)
16:46:16.136 -> Connecting to AP 38:10:D5:1A:64:0A CH: 11
16:46:16.681 -> WiFi begin...
16:46:16.681 -> Timing Net: 104ms
16:46:34.620 -> (evt) Disconnected from WiFi STA access point. Connecting: 1
16:46:38.853 -> (evt) Completed scan for access points (3)
16:46:38.853 -> Connecting to AP 38:10:D5:1A:64:0A CH: 11
16:46:39.402 -> WiFi begin...
16:46:39.402 -> Timing Net: 104ms
16:46:42.188 -> (evt) Disconnected from WiFi STA access point. Connecting: 1
16:46:46.378 -> (evt) Completed scan for access points (3)
16:46:46.378 -> Connecting to AP 38:10:D5:1A:64:0A CH: 11
16:46:46.856 -> WiFi begin...
16:46:46.856 -> Timing Net: 104ms
16:46:49.715 -> (evt) Disconnected from WiFi STA access point. Connecting: 1
16:46:53.922 -> (evt) Completed scan for access points (3)
16:46:53.922 -> Connecting to AP 38:10:D5:1A:64:0A CH: 11
```

rstrouse, 2024-08-14:
Is the passphrase correct?

DrmedAllel, 2024-08-14:
Yes it is. I have tried it multiple times. Also changed the board...

rstrouse, 2024-08-14:
Do you have any special characters or spaces in the SSID?

DrmedAllel, 2024-08-14:
SSID is "Wi-Fi-Teck12".
Sometimes I get the connection to the wifi but this is very rare. Most of the time I am stuck in the loop...

rstrouse, 2024-08-14:
I don't think the dashes should be a problem.  It is very odd that it looks like the router is responding with the connection but it is immediately disconnecting.  It is scanning 3 APs do you have a mesh?

DrmedAllel, 2024-08-14:
Yes I do. There are three repeaters and one router in my network.

DrmedAllel, 2024-08-15:
Alright, I will do so. Thank you for the suggestion. I will give you feedback once I‘ve tried.

DrmedAllel, 2024-08-18:
I've tried it with disabled roaming but still wasn't to reach the web-interface after the setup process. Where can I change the setting for the hidden SSIDs?

DrmedAllel, 2024-08-18:
After put in the SSID & pasword I click on "save adapter", then the logs start looking very strange:

`22:31:06.218 -> Socket [0] Connected from 192.168.4.2 url: /
22:31:06.218 -> Initializing SocketLoading file /index.html
22:31:13.570 -> Timing WebServer: 1413ms
22:31:13.570 -> Socket [0] Disconnected!
22:31:13.642 -> Socket [0] Connected from 192.168.4.2 url: /
22:31:13.642 -> Initializing Socket Client 0
22:31:13.714 -> Max Heap: 110580
22:31:13.714 -> Free Heap: 164396
22:31:13.714 -> Min Heap: 145960
22:31:14.513 -> Client 0 joining room 0
22:32:28.137 -> Cancelling WiFi STA Scan...
22:32:28.137 -> Cancelling WiFi STA Scan...
22:32:28.137 -> Cancelling WiFi STA Scan...
22:32:28.137 -> Cancelling WiFi STA Scan...
22:32:28.137 -> Cancelling WiFi STA Scan...
22:32:28.137 -> Cancelling WiFi STA Scan...
22:32:28.137 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.166 -> Cancelling WiFi STA Scan...
22:32:28.204 -> Cancelling WiFi STA Scan...
22:32:28.204 -> Cancelling WiFi STA Scan...
22:32:28.204 -> Cancelling WiFi STA Scan...
22:32:28.204 -> Cancelling WiFi STA Scan...
22:32:28.204 -> Cancelling WiFi STA Scan...
22:32:28.204 -> Cancelling WiFi STA Scan...`

rstrouse, 2024-08-20:
That is very strange.   Reboot the device by removing power then let it boot back up again.

hmbacher, 2024-09-18:
Hi guys!
Unfortunately I am facing the exact same issue. Only in rare cases the module successfully connects to my TP-Link access points mesh after startup. 

**Hardware**: XIAO ESP32S3 **Firmware+App**: 2.4.6

**Serial interface logs**:
```
Startup/Boot....
Mounting File System...
File system mounted successfully
Chip Model ESP32-s3
Preference IP Free Entries: 493
SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
Connection Type: 1
NTP Settings 
pool.ntp.org TZ:CEST-1CET,M3.2.0/2:00:00,M11.1.0/2:00:00
WIFI Settings
SSID: [AztecNet-Gen2] PassPhrase: [***]
Launching web server...
Creating Web MicroServices...
WiFi Mode: 0
(evt) WiFi interface ready
WiFi station mode started
Socket Server Started...
App Version:2.4.6
shades.cfg exists so we are using that
Reading header at 0
version:24 len:76 roomSize:29 roomRecs:1 shadeSize:276 shadeRecs:1 groupSize:200 groupRecs: 0 pos:76
Setting S3 Transceiver Defaults...
Applying radio settings Setting Data Pins RX:7 TX:3
Setting SPI Pins SCK:4 MISO:6 MOSI:8 CSN:9
Radio Pins Configured!
Successfully set up the radio
Enabled receive on Pin #7 Timing: 1
Initializing RX Queue

(evt) Completed scan for access points (5)
Connecting to AP 50:C7:BF:10:6C:16 CH: 1
WiFi begin...

Timing Net: 107ms
(evt) Disconnected from WiFi STA access point. Connecting: 1
(evt) Disconnected from WiFi STA access point. Connecting: 0
(evt) Completed scan for access points (5)
Connecting to AP 50:C7:BF:10:6C:16 CH: 1
WiFi begin...

Timing Net: 102ms
(evt) Disconnected from WiFi STA access point. Connecting: 1
(evt) Completed scan for access points (5)
Connecting to AP 50:C7:BF:10:6C:16 CH: 1
WiFi begin...

Timing Net: 102ms
(evt) Disconnected from WiFi STA access point. Connecting: 1
(evt) Completed scan for access points (5)
Connecting to AP 50:C7:BF:10:6C:16 CH: 1
WiFi begin...

Timing Net: 102ms
(evt) Disconnected from WiFi STA access point. Connecting: 1
(evt) Completed scan for access points (5)
Connecting to AP 50:C7:BF:10:6C:16 CH: 1
WiFi begin...

Timing Net: 102ms
(evt) Disconnected from WiFi STA access point. Connecting: 1
(evt) Completed scan for access points (5)
Connecting to AP 50:C7:BF:10:6C:16 CH: 1
WiFi begin...

Timing Net: 102ms
(evt) Disconnected from WiFi STA access point. Connecting: 1
(evt) Completed scan for access points (5)
Connecting to AP 50:C7:BF:10:6C:16 CH: 1
WiFi begin...
```

**Some additional information**:
It worked quite stable a while ago (continuously connected to WiFi for weeks) and direct WiFi re-connection on almost any startup. _Now_, the system crashes after a couple of hours with stack overflow issues und does not successfully reconnect:

```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (async_udp) 
Core  0 register dump:
PC      : 0x420be5b0  PS      : 0x00060236  A0      : 0x820bab44  A1      : 0x3fcb8400  
A2      : 0x3fc9a8ac  A3      : 0x00000010  A4      : 0x00000010  A5      : 0x3fcb8410  
A6      : 0x00000001  A7      : 0x0000000c  A8      : 0x00000001  A9      : 0x3fcb8410  
A10     : 0x0000005a  A11     : 0x0000002a  A12     : 0x00000073  A13     : 0x3c101400  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
Backtrace: 0x420be5ad:0x3fcb8400 0x420bab41:0x3fcb8710 0x4200b6d7:0x3fcb87d0 0x4200bd02:0x3fcb87f0 0x4200c5b5:0x3fcb8e20 0x4200c649:0x3fcb9190 0x420309d9:0x3fcb91b0 0x42030e75:0x3fcb91e0 0x42030eca:0x3fcb9280
```

This is why I started thinking this might be a HW issue as well. So I am doing two things now:
1) I deactivated the Roaming feature and I will observe the system for any improvement
2) I ordered some new XIAO S3 modules to check the behaviour off different modules of the same type.

I will keep you updated with my impressions.
**AWSOME project, by the way!**

hmbacher, 2024-09-28:
Hi,
here the updates as promised:

1) Romaing feature does not have an effect on the issue.
2) A brand new XIAO ESP32S3 module works fine, if just connected via USB-C port but shows the same issue if mounted to the actual target PCB.
3) XIAO ESP32**_C3_** module works perfectly, either isolated (just via USB) or integrated (put to the target PCB).

I assume it is not (only) a software issue. To identify the actual root cause, further low level analysis is required. As I am flexible in using S3 oder C3 type, I have working and stable solution now with using the C3-module.
Best regards!

theincogtion, 2024-10-10:
I am using an AVM Mesh Network and I have regular problems with the connection. Most of the time it connects after start but then after some hours or days the connection gets lost.
Sometimes I have to reboot several times. Please let me know if you need further details.

diegoflyer, 2025-03-16:
@rstrouse I don't know if it is the same issue but I try to explain here.
After installing the firmware on my wt32 the wifi access point is not available.
The board is ok because it works with esphome in wifi mode.
I tried with radio antenna connected and not connected, must the antenna be connected on first startup after firmware installation?
Thanks

diegoflyer, 2025-03-16:
@rstrouse  here what I have from serial log:
`20:31:12.697 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:31:12.697 -> flash read err, 1000
20:31:12.697 -> ets_main.c 371 
20:31:13.079 -> ets Jun  8 2016 00:22:57
20:31:13.079 -> 
20:31:13.079 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:31:13.079 -> flash read err, 1000
20:31:13.079 -> ets_main.c 371 
20:31:13.436 -> ets Jun  8 2016 00:22:57`

diegoflyer, 2025-03-16:
Sorry @rstrouse my mistake, I was uploading the wrong firmware (not the onboarding one)