rstrouse/ESPSomfy-RTS #454: Can't get cC1101 to work according to the Mini Guide

mwitkow, 2024-08-17:
### Hardware

ESP32-C3

### Firmware version

v2.4.6

### Application version

v.2.4.6

### What happened? What did you expect to happen?

I successfully built a device using a WROOM ESP S2 and a CC1101 (as described in this [wiki guide](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device)), and wanted to build a smaller version of it (with a detachable antenna) using the Seed Studio C3 (as described in this [wiki guide](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Mini-ESP32s3-Device-Build).

I successfully flashed the device using `SomfyController.onboard.esp32c3.bin` and it starts up without the CC1101 radio attached (the WiFI fallback AP appears). However, when attaching the radio as [pinned out in the guide](SomfyController.onboard.esp32c3.bin), the WiFi doesn't come up. 

It does come up when wiring the TX and RX pins to D6/D7 pins (see [pinout](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/#pinout-diagram)), but their respective GPIO numbers (GPIO21, GPIO 20) are not available in the drop down lists. 

There are no logs on serial that are available.

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-08-18:
The C3 and the S3 are two distinctly different chips.  Also, you do not want to use GPIO20 and GPIO21 for this.  Those pins are mapped to the serial UART on the board.  Try the pinout below.

| Pin | Color | ESP32 | Description |
| --- | ----- | ----- | -------- |
| 1 | Black | GND | Ground |
| 2 | Red | 3v3 | VCC |
| 3 | Green | GPIO3 | GDO0 - TX Pin |
| 4 | Blue | GPIO6 | CSN |
| 5 | Orange | GPIO8 | SCK |
| 6 | Purple | GPIO10 | MOSI |
| 7 | White | GPIO9 | MISO |
| 8 | Yellow | GPIO4 | GDO2 - RX Pin |

mwitkow, 2024-08-19:
That's precisely the pinout I have.
[image]

What I'm observing is that: 
 without the radio attached, the WiFi AP (EspSomfy) comes up
 with the radio attached as per pinout, there's no WiFi AP

in either case I'm unable to get logs from the device via USB serial to provide any further logs.

I tried with different radios (got 2) but only one board. Is there any way to get logs of the board? or should I buy another board assuming this one's broken?

rstrouse, 2024-08-19:
I assume you are using the same cable that you used to flash the device from the ESPHome website which would verify that it is a data cable.   After it flashes can you connect to it from the ESPHome website or is there no comm port identified?   I will have a look to see if there is some setting for the Seeed that is different on the compile since I have a couple of these to try.

mwitkow, 2024-08-19:
Yeah, it's a data cable. I used it to flash a bunch of things already including the Seeed C3 (which boots to Wifi without radio), as well as a WROOM-32U. Compared to the WROOM-32U, which had logs both in ESPHome logs as well as `screen /dev/cu.usbmodemXYZ`, I can't get anything out of a Seeed C3.

rstrouse, 2024-08-19:
I'll bet there are jtag settings for the serial out on the compile line for that board.  I am pretty sure I have a Seeed C3 in my stash.

rstrouse, 2024-08-19:
I added a compile flag to disable jtag and enable serial output on some models of the C3 please download the SomfyController.onboard.esp32c3.zip file from the v2.4.7 pre-release.
https://github.com/rstrouse/ESPSomfy-RTS/releases/tag/v2.4.7

Unzip this file and flash with the resulting SomfyController.onboard.esp32c3.bin file.

mwitkow, 2024-08-21:
The logs say:

```
[09:21:02]
[09:21:02]Startup/Boot....
[09:21:02]Mounting File System...
[09:21:02]File system mounted successfully
[09:21:03]Chip Model ESP32-c3
Preference IP Free Entries: 621
SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
[09:21:03]Connection Type: 1
NTP Settings 
[09:21:03]pool.ntp.org TZ:
[09:21:03]WIFI Settings
[09:21:03] SSID: [] PassPhrase: []
[09:21:03]
[09:21:03]Launching web server...
[09:21:03]Creating Web MicroServices...
[09:21:04]WiFi Mode: 0
[09:21:04](evt) WiFi interface ready
[09:21:04]WiFi station mode started
[09:21:04]Socket Server Started...
[09:21:04]App Version:2.4.7
Starting clean
[09:21:04]Initializing RX Queue
[09:21:04]
[09:21:04]Turning the HotSpot On
[09:21:04](evt) WiFi SoftAP Started IP:192.168.4.1
[09:21:04](evt) WiFi SoftAP Stopped
[09:21:04](evt) WiFi SoftAP Started IP:192.168.4.1
```

So it seems things work with 2.4.7 and the AP is coming up, as well as the radio (tried both). Thanks for the new build, no idea what changed but that kind of fixed it.

Windyo, 2025-09-22:
@rstrouse I've tried finding what you changed but no luck - I'm trying to communicate with a cc1101 from a c3 supermini and getting the same sort of error https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/170 - seeing as your code does in fact run, and it's the same hardware, do you have an idea of what's happening ? I know this isn't really linked to your project, but I appreciate your guidance.