rstrouse/ESPSomfy-RTS #455: Blind not working when at top position - works when moved to any other position with original remote

Wishmaster77, 2024-08-17:
Hi,
i had to redo my config and have everything working again. 
However one of the blinds now doesnt react to any command from ESPSomfy-RTS when in the topmost position.

As soon as i move it to any other position with the original remote, e.g. down for 1 sec, i can again move it as i want via ESPSomfy-RTS - until i move it to the topmost position again.
The original remote works fine all the time as it should... 

Any Idea what could cause this? Is there anything i can do to troubleshoot?

Thanks,
Sebastian

rstrouse, 2024-08-18:
The most likely cause of this is a sleeping motor.  Add a couple of command repeats to the ESPSomfy RTS configuration for that blind.

c1328, 2024-08-25:
Gracias! That helped me :-)