rstrouse/ESPSomfy-RTS #456: Roller Shutter landing/takeoff time

ghisch, 2024-08-18:
### Describe you new feature you'd like

Hello,,

on outdoor roller shutter ([like this](https://www.neuffer-windows.ca/sites/ca/files/img/rollershutters/front-mounted-rollershutter-angular.jpg)), the open percentage computed from open/close duration is not accurate because there's 2 closed states:
1. [Partially closed](https://www.shutterstock.com/shutterstock/videos/1050656401/thumb/1.jpg?ip=x480): when the shutter touches the ground but is expanded (some light going through)
2. [Fully closed](https://www.qsec.co.uk/wp-content/uploads/2019/07/lightweight-shutter-1.jpg): when the shutter touches the ground and is fully contracted (no light going through)

I defined the close duration from _fully open_ to _fully closed_, the open duration from _fully closed_ to _fully opened_.
Let say my close duration is 10s. From an fully opened shutter, If I set shutter to 50%, the shutter will close for 5s then stop. But 5s of closing is not a 50% open shutter, it's more something like 30% open shutter. This issue does not appear with Shades, only happens with roller shutter.

It's nothing critical, just a nice to have. I have some ideas to implement that:
1. Do not use a linear curve to compute open percentage of a shutter from open/close duration. Instead, use log or bi-linear curve
2. Implement landing/takeoff time in addition of open/close duration.

Let me know what you think of it, or if I missed an obvious way of achieving this.

Thanks for ESPSomefy-RTS btw.

rstrouse, 2024-08-18:
I have a few other mechanisms that when I get some time, I may look at.  These are akin to the integrated tilt mechanisms that appear on blind motors which ESPSomfy RTS supports very well.  Essential the motor is still linear but has a mechanical means of performing another action at the extremes of the motor.

To get a better idea of what I am talking about look at the settings for an integrated tilt motor.   These use slat rotation extremes to perform the lift.
[image]

Nickduino, 2024-11-13:
@ghisch I don't think how you could improve that. I want my shutters to be partially closed, sometimes. Maybe a custom shutter card in HA could, though: it would **show** the shutters fully closed vs partially open and at about half the height when they're actually open at 60%