rstrouse/ESPSomfy-RTS #457: Home Assistant recognizes the new configured ESPSomfy RTS but all entities are unavailable.

kibeki, 2024-08-18:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

After I successfully configured the ESPSomfy RTS for my 4 blinds, I tried to integrate the device into Home Assistant.
Home Assistant can see the configured blinds, but is not able to control them.

When I open a blind Home Assistant shows "This entity is unavailable."

Is this behavior something that is already know and you can give me a tipp to solve my problem?

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-08-18:
Try restarting Home Assistant and refreshing your browser cache.  If that does not fix it then check the IP address configured for the plugin actually matches the IP address of the ESPSomfy RTS device.   

TIP: If you haven't yet reserve the IP and host address in your router DCHP.

kibeki, 2024-08-18:
After restart of my HA the controls up/down/stop works. 

Is there any chance to use the "My" -  position function through HA?

rstrouse, 2024-08-18:
Yes if the motor is at rest it will move to my position if you hit the stop button.