rstrouse/ESPSomfy-RTS #459: Setting blinds to My position isn't recognized

phatina, 2024-08-20:
### Hardware

ESP32

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

If I have my blinds in any other position than "My position" and press My button on Telis 4 remote, they go down to that default position (completely down and 45° tilt). ESPSomfy doesn't pick this command and it thinks that the blinds are open. Am I missing any configuration? Thank you for your time and opinions.

### How to reproduce it (step by step)

```bash
1. Set Somfy blinds to arbitrary position,
2. Press "My" button on a remote
```

rstrouse, 2024-08-20:
Are you sure your blind motors support the My position if the do not click the Simulate Favorite checkbox.

phatina, 2024-08-20:
Simulate Favorite Position is unchecked on all of my blinds.

rstrouse, 2024-08-20:
Do you have a my position that is set in ESPSomfy RTS and if so when you set it did the motor jog?

phatina, 2024-08-21:
No, I don't have My position set in ESPSomfy yet. Do I need to do it so that ESPSomfy picks up My button press on a remote? Thank you.

rstrouse, 2024-08-22:
Yes

https://github.com/rstrouse/ESPSomfy-RTS?tab=readme-ov-file#setting-a-favorite

llaumgui, 2024-11-20:
Hi,

same issue with my pergola. I haven't state feedback. I would like to use up / My / down without any position restriction.

rstrouse, 2024-11-27:
ESPSomfy RTS adds the position feedback to your motor by setting the motor timings.