rstrouse/ESPSomfy-RTS #465: Venetian blind moves independently

011V32, 2024-09-02:
### Hardware

LilyGO T-Internet POE

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

I don't know why, but today one of our venetian blinds simply closed itself step by step.
In HomeAssistant I can see that it wasn't done by anything else, but by the module itself.
Can you find out what the problem was?
[image]
[image]

### How to reproduce it (step by step)

```bash
Unfortunately, I can't say. The system has been online for many weeks and this is the first time it has done this...
```

rstrouse, 2024-09-04:
There really isn’t anything that would trigger this in ESPSomfy RTS unless it is requested to do so from some other source.  Are you sure there is nothing setting the position over the rest api.

011V32, 2024-09-04:
I only use HomeAssistant in conjunction with this device. My family then uses the device via HomeKit, which is provided by HomeAssistant. So directly or another device such as HomeAssistant does not connect to ESPSomfy-RTS. The commands were not executed by HomeAssistant, as you can see from the log. When I then restarted the device, this phenomenon stopped again. I can see whether it occurs again in a few weeks...

rstrouse, 2024-09-04:
HA does not really execute any commands.  However, it is capable of sending a rest request to execute the command to ESPSomfy RTS.   Are you sure that it is sending a STEP command?   You can see this in the Last cmd attribute for the entity as well as the source of the command.  If you have a wheel that is spitting these out it should tell you which remote is doing it.  Bear in mind that ESPSomfy RTS does not issue STEP commands unless it is told to do so externally.   All control from the device will be UP/DOWN/MY commands to move the motor from the interface but it will hear those commands and adjust the position accordingly.

011V32, 2024-09-07:
If the command is executed with HomeAssistant, it says with which integration etc. it was executed by. Only if it is not executed by/through HomeAssistant, as in the image above, nothing is said. Since I didn't see anything in the logs, I assume that it was executed by ESPSomfy-RTS.
But since it doesn't occur again after the restart, I can't diagnose it at the moment.
What should I look out for or what should I do if it happens again?
Thanks.

011V32, 2024-11-10:
Since it hasn't happened again so far, I'm going to close it for now.
If it happens again, I'll reopen it.

closed 2024-11-10 by 011V32