rstrouse/ESPSomfy-RTS #466: Sun and warning icon?

XilviuX, 2024-09-02:
### Hardware

Other

### Firmware version

v.2.4.6

### Application version

v.2.4.6

### What happened? What did you expect to happen?

I'm trying to follow the Somfy instructions to fine-tune the higher and lower end position of the curtain. I can't seem to make this work, but during the different tests I saw that the web interface displays these icons. What is their meaning?
[image]

### How to reproduce it (step by step)

```bash
1. follow instructions from Somfy to adjust the high and low positions (which don't work for my curtain anyway, yay), that is: hold up and down together for 5 seconds, adjust the position, and press "my" for 2 seconds.
2. the icons appears in the interface
```

rstrouse, 2024-09-04:
Please go to the virtual remote, select the motor from the drop-down and send a sensor command to the motor.  That will clear the sunny and windy sensor settings.

XilviuX, 2024-09-04:
Just tried it and it didn't change anything. I also tried setting, then unsetting the checkboxes at the bottom of the virtual remote page (sunny and windy sensors) and this worked - the icons are gone. Thanks for the tip.