rstrouse/ESPSomfy-RTS #469: Somfy Gate 2 Buttons remote

aidechoute, 2024-09-09:
### Describe you new feature you'd like

Hello,
First of all thanks for your wonderfull work!!! It's amazing
Here is my "issue"
I have configure the ESPSomfy RTS for my shade without any problem. I also have a Somfy Exavia 500 Gate that I want to configure.
With this gate I have when i bought it 2 preconfigure remote with only 2 buttons
-Button 1 open the gate totally.
-Button 2 open just one of the 2 gate for pedestrian access.
Here is a link to the manual (sorry it's in french: https://www.somfy.info/download_sheet/5073786/5073786B_EXVIA%20SGS%20SAV.pdf )

The sync process for the gate is easy just put the remote on a target on the motors then click on button 1. This will configure button 1 and button 2 automaticly.

In ESPSomfy I do not find remote with 2 Button. I have configure my gate with the 3 button remote (the one with "My") it's working but only for the "total opening". I also try the "dry contact' button without more luck.

If you have any idea how to configure it could be fantastic 🗡️ 

Thanks

rstrouse, 2024-09-09:
I believe there is a setting on the Exavia to support the 3 button remote and the two button remote you have is actually two 1 button remotes with a unique address per button.   We can verify this by looking at the log when you press each button.  If the address is different for each button then this is the case.

aidechoute, 2024-09-09:
Yes buttons are different
[image]
In blue pedestrian and in red total open

aidechoute, 2024-09-09:
How to create 2 definitions? (I'm a noob :D )

aidechoute, 2024-09-09:
In the log:
In red it's button1 and blue is button2
(Really appreciate your help)

rstrouse, 2024-09-09:
Capture that again but this time please press the copy button then paste those results here.

aidechoute, 2024-09-10:
[Exavia.txt](https://github.com/user-attachments/files/16940503/Exavia.txt)
address 9690619 is the pedestrian opening
address 9625083 is the total opening

rstrouse, 2024-09-10:
Well look at that.  There is nothing really special about the frames here.  But I have a hypothesis on this.

### Assumptions
I have come up with some assumptions based on your log.  I feel pretty strongly about how this works.  I know that the address on every channel on every remote I have seen has been offset by 65536.  So each channel address on every remote is 
`channel 1 address + ((channel number - 1) * 65536)`

**Assumption 1:**
If an address is paired into the Exavia to move the entire gate it will reserve the address for the pedestrian access as <gate address> + 65536.

Each address on a 2 button or 4 button gate remote is 65536 larger than the next button so
9625083 = Button 1
9690619 = Button 2
If there were a button 3 and 4
9756155 = Button 3
9821691 = Button 4

**Assumption 2:**
If you add a shade in ESPSomfy RTS with a starting address of 4824452 this would be button 1 then add 65536 that address for button 2 to have 4889988 for the button 2 address you will have made a virtual 2 button remote in the Somfy world.

### Steps to make this work

**Step 1:** Unpair any ESPSomfy RTS addresses that you have with the Exavia
**Step 2:** Create a Gate (1-button center) shade definition in ESPSomfy RTS and assign it a remote address of **4824452**.
**Step 3:** Pair this to the the Exavia and it should pair to open and close both panels.
**Step 4:** Create a Gate (1-button left) shade definition in ESPSomfy RTS and assign it a remote address of **4889988**.  The address in this instance is critical because Exavia will look at this address and put it in the pedestrian slot to only move one motor.
**Step 5:** Pair this to the Exavia and it should pair to the pedestrian slot.

aidechoute, 2024-09-11:
I have try your recommandation
Create 2 different shade with different remote address
[image]
[image]
Unfortunately when I try to pair it nothing happen

rstrouse, 2024-09-11:
Please change the bit length for those two definitions to 80-bit and try again.

If that does not work try the method below.

Try this:
1. Press and hold the power button on the motor for 2 seconds.  The power light should come on
2. Make sure the ESPSomfy RTS device is near the motor.
3. In ESPSomfy RTS open the Virtual Remote tab and select the motor from the dropdown.
4. Press the toggle button there.

aidechoute, 2024-09-12:
So with a 80-Bits I can pair the remotes but still the same issue it only open the full gate and not the pedestrian

And regarding the other method I do not have power button on my motor :(

rstrouse, 2024-09-12:
Are you sure you set the addresses as indicated above?

rstrouse, 2024-09-12:
Alright so I have another thing we can try.  It appears all of the keytis style remotes start at a higher address.

### Steps to try this scenario
**Step 1:** Unpair each of the shade definitions that you previously paired for the 2 shade definitions.
**Step 2:** Change the address of the shade definition that is to open both panels to `10663760` then pair it.
**Step 3:** Change the address of the shade definition that is to open only the pedestrian gate to `10729296` then pair it.
**Step 4:** Test it and let me know if it is successful.

aidechoute, 2024-09-13:
I will try on sunday and let you know...I'm not at home
Thanks for your patience and your help

Remitooo, 2024-09-22:
hello !
how do you pair the gate motor in espsomfy ? 
i have a  evolvia gate motor and i dont understand how i can pair

aidechoute, 2024-09-23:
@Remitooo 
First try to open it with your remote and check in log if your ESPsomfy see something (It should as it's an RTS motor)
Then create virtual remote in espsomfy with a 80bits. then pair it with your espsomfy
In my case i have to open the motor and put the antenna of my ESPsomfy on the target in the motor then click pair

jc-m, 2024-11-26:
@rstrouse could you describe how you arrived at the values to use for the two remotes ? I see that they are both separated by 65536 like before but using a different starting point. I have an Evolvia 400 with two remotes (not yet the ESPsomfy - hardware is ordered), and would like to understand the logic so that I can use your software on this gate.