rstrouse/ESPSomfy-RTS #470: Enhanced remote funktion and SX1276 option.

palsbo, 2024-09-12:
### Describe you new feature you'd like

First: thank you for a really good and well-worked project.
I have implemented it with HA for 3 awnings and 2 slat curtains. For the latter, I have a Remote model Situo 5, which in addition to "Up", "Down" and "My" buttons also has a wheel around the "My" button. This ring is especially useful for blinds. It sends the code 0x0B, which has not normally been used for Somfy Remotes as far as I know. I don't know if it might be a consideration to implement it in your outstanding project.
With my old man's eyes, I'm better at software than hardware, and I had a lot of trouble connecting a Wemos ESP32 Mini D to the CC1101 module.
I have previously done a similar project where I have used a Lilly Lora T1 433Mhz module. It has everything gathered in one module including an oled display, and there are 3D files for built-in boxes available on the web. A good overall solution. It use SX1276 instead of CC1101. Could it be a consideration to make this a possible alternative?..

rstrouse, 2024-09-12:
I have looked at the SX1276 and do have plans to remove the Elechouse libraries.  It is on the list.   However, ESPSomfy RTS already supports all of the 80-bit command set including Step Up/Down with the ability to send the step size.  This is only available when the motor is defined as an 80-bit RTS.

[image]

palsbo, 2024-09-12:
Thanks for quick reply.
I was not aware of the 80-bit option - looks great.
I would look forward to the SX1276 solution.
Best regards
Erik

image.png (view on web)<https://github.com/user-attachments/assets/0c4873ad-3f27-4045-8949-b9a7570c340a>