rstrouse/ESPSomfy-RTS #471: PHY Chip Type W5500

till1978, 2024-09-21:
### Describe you new feature you'd like

Hello
great work!
Is there any way to integrate the W5500 Ethernet module?
My ESP32 is a LilyGO-T-ETH-Lite
Thanks

rstrouse, 2024-09-22:
Unfortunately, the W5500 chip is not supported in the Arduino core version being used.  This will have to wait until I have time to move to core 3.x.  Unfortunately, the compiled code ends up being too big for the partition at the moment.

till1978, 2024-09-22:
Thanks for the super quick answer.
I ordered another board.
Thanks again for the great work. 
Is there an opportunity to give a tip?

SniffeuR, 2025-09-03:
bonjour,
j'ai moi aussi acheté une carte WaveShare ESP32-S3-ETH équipée d'une puce wifi W5500 (je ne vois votre réponse qu'après l'avoir reçue) avec son module POE et son grand nombre de GPIO disponibles.
Pensez-vous réussir à l'intégrer un jour ?

Spokplacenta, 2026-01-18:
@SniffeuR je viens de l'intégrer, ça faisait un moment que ça m'embettait d'utiliser cette carte en Wifi.
J'ai fais un fork https://github.com/Spokplacenta/ESPSomfy-RTS si tu veux essayer en attendant que la [PR ](https://github.com/rstrouse/ESPSomfy-RTS/pull/633) soit acceptée.