rstrouse/ESPSomfy-RTS #472: Paring Gate motor

Remitooo, 2024-09-22:
Hi 
first  this project is amazing 

i have all my rolling shades connected perfectly 
now i'm trying to connect my gate motor 
it is an evolvia 400 with keytis ns2 remotes 

the problem is the remotes don't have the prog button ,
to pair a new one normaly i just place it on top off the motor circuit bord and i press 1 button off the remote 
the led blink i press the button an other time and its done 

i've try with my remote to put the motor in paring mode but when i press prog in espsomfy nothing happen 
i have selcect 80bits ( thats what i saw in the log when i press the button off my remote)

thank you in advence

rstrouse, 2024-09-23:
Select 80 bit one button gate and it will send the proper toggle command on pairing.

Remitooo, 2024-09-23:
Yes this is what i have done 

[image]

But how i put the motor in pairing mode ?

rstrouse, 2024-09-23:
I believe the instructions say to hold the button that looks like a power button until it starts to flash.