rstrouse/ESPSomfy-RTS #474: Change WiFi SSID -or- force access point?

nicoodeimos, 2024-09-30:
### Describe you new feature you'd like

Hello there, I'm sorry if this has already been answered before, but I couldn't find any reference to this.
Basically, if I want to change the configured SSID before my Wifi name has changed, is there any way to return to access point mode to be able to change settings?
Many thanks!

rstrouse, 2024-09-30:
You can go into the network tab and remove the SSID then press save.

nicoodeimos, 2024-09-30:
But what if the esp device is unjoinable? (for ex. I move into a new house with a new SSID, and the ESP doesn't know the new environment).

rstrouse, 2024-09-30:
It will automatically fall back to the Soft AP mode if it cannot find the SSID after 20 seconds.

nicoodeimos, 2024-09-30:
Awesome ok, thank you so much!

closed 2024-09-30 by nicoodeimos