rstrouse/ESPSomfy-RTS #475: Can't seem to have any RSSI

nicoodeimos, 2024-10-01:
### Hardware

ESP32-C3

### Firmware version

2.4.7

### Application version

2.4.7

### What happened? What did you expect to happen?

So, I have ordered those two cards and proceeded to connect everything:
- https://fr.aliexpress.com/item/1005006391993583.html?spm=a2g0o.order_list.order_list_main.10.21ef5e5bSitY9x&gatewayAdapt=glo2fra
- https://fr.aliexpress.com/item/1005006427924521.html?spm=a2g0o.order_list.order_list_main.5.231a5e5b9wErPR&gatewayAdapt=glo2fra

I successfully configured the radio (the banner has disappeared) but can't seem to have any RSSI shown. 
Is that normal? I'm not sure what to test, I have verified 30 times the PIN on the board.

[image]

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-10-01:
Which remote do you have, are you sure it is RTS?

If you have an RTS (not IO) remote then this is most likely incorrect wiring.  Which esp32 pin is connected to pin 3 GDO0 of the transceiver?

nicoodeimos, 2024-10-01:
So I don't have any remote yet, not even a blind (I'm preparing the ESP device at the moment).
I have ordered a remote though to test, that has the same characteristics as the final site where it will be installed (I don't have access to the physical place yet).

I don't think it is an RTS one, but I suspect it is an RTV-L at 433.92 MHz.
Is that normal that no RSSI is shown if I no signal has been sent/received? It doesn't mean my wiring is wrong?

GPIO 3 of my ESP32-C3 module is connected to PIN 4 of the transceiver.

rstrouse, 2024-10-01:
Oh so you are not sending any traffic because you do not have a remote.  You will not see an RSSI if there is no signal to capture.  When you get the remote you should be able to see an RSSI whenever a button is pressed on the remote.

nicoodeimos, 2024-10-03:
I have received the remote today, I'm not receiving any traffic. The logs and RSSI seem to stay blank.
I'm guessing the protocol is not RTS/RTL-RTV.

The frequency (written on the back of the remote) is 433.92 and it seems to use "52 bit FLOR + rolling code" protocol, whatever that means.

The model is the following: [Nice Era MW1](https://clic-volet.fr/commande-nice/921-telecommande-nice-radio-portable-era-mw1-miniaturise-1-canal.html?gad_source=1&gbraid=0AAAAA9XBeD6JVGm1ScKnjVitl3uePj-eq&gclid=CjwKCAjwgfm3BhBeEiwAFfxrGxW4HEIzfZ4Ea23JfsbBQZ9IQKZNJNNHsdDPZA3lUfYmV1L_H2EoUxoCRAAQAvD_BwE).

I have set the base frequency on the firmware to 433.922 (can't seem to be able to set it precisely to 433.92 with the slider).
Is there any way to confirm the protocol is not supported?

[image]

Many thanks!

rstrouse, 2024-10-04:
Unfortunately that protocol is not supported.  I looked to see if there is some documentation on the timings but I didn’t find anything.