rstrouse/ESPSomfy-RTS #476: Can't receive and transmit at same time

davidglvn, 2024-10-05:
### Hardware

Olimex ESP32-PoE/EVB

### Firmware version

2.4.6

### Application version

2.4.6

### What happened? What did you expect to happen?

First thanks for the great work!
Second, I'm don't think it's a bug and probably hardware issue, but I'm not able to make it to receive and transmit at the same time. I've followed the guide and connected everything on Olimex ESP32-PoE-ISO with E07-M1101D(from aliexpress) antenna, and had an issue receive at first and only after switching RX and TX in the incorrect way I was able to receive commands (after reading the issues here I've tried it):
Pin | Description | ESP Pin
-- | -- | --
1 | GND | GND
2 | VCC | 3v3
3 | GDO0 - This is the TX Pin | ~GPIO 04~ **GPIO 36**
4 | CSN | GPIO 13
5 | SCK | GPIO 14
6 | MOSI | GPIO 15
7 | MISO | GPIO 16
8 | GDO2 - This is the RX pin | ~GPIO 36~ **GPIO 04**

During the pairing with the motor it didn't worked of-course, swapping them back allowed the motor to pair, but it's not able to receive commands from the remote now. I'm not sure what else I can debug to understand where is the issue and what to replace. If you have any ideas what can be the issue or how to debug to get more info I'll be glad....

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-10-06:
Typically GPIO36 on the ESP32 is an input only pin and cannot be used for output.  Connect GPIO4 to GDO0 (pin on the transceiver) and set **both** the TX and RX settings in the configuration to GPIO4.

davidglvn, 2024-10-07:
@rstrouse thanks works like charm! 
do you think I can assume it's antenna issue and order a new one? or this can be stable solution?

rstrouse, 2024-10-07:
You are using a stable solution.  ESPSomfy RTS is multiplexing the connection with the configuration.   At any given time there can only be one valid frame on the frequency at at time anyway and the transceiver is only ever transmitting or receiving,  It will never do both at the same time because there is only one channel.

davidglvn, 2024-10-15:
got you!
thanks for the help and for the great module!

closed 2024-10-15 by davidglvn