rstrouse/ESPSomfy-RTS #477: Olimex ESP32-POE / E07-M1101S-SMA

Qualitrade, 2024-10-09:
### Hardware

Olimex ESP32-PoE/EVB

### Firmware version

v.2.4.7

### Application version

v.2.4.7

### What happened? What did you expect to happen?

Connected the E-07 one to one to the Olimex Ext ports. 1-1, 2-2, 3-3, 4-4, and so on
In the esp radio configuration setting i used TX=GPIO04, CSN=GPIO13, SCLK=GPIO14, MOSI=GPIO15, MISO=GPIO16 and RX+GPIO36.

when saving these setting and enabling the radio i get an 500:service error, service put/save radio.
when unchecking the enable radio then saving works. but when enabling radio and saving setting the same error appears.
what is wrong

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-10-09:
Please ignore the pin numbers on that board and only use the pinouts as defined by their GPIO counterparts.  You do not match pin 1 on the Olimex to pin 1 on the transceiver.  You will find a mapping of the pins at the following url.

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/resources/ESP32-POE-ISO-GPIO.png

Also, if you are using the UEXT connector on the board make sure you pay close attention to the orientation of the pins.  You will see the key slot to the left in the diagram.  

The board is likely rebooting when it tries to activate the pins so either you have mis-wired on the transceiver or on the Olimex.

Qualitrade, 2024-10-10:
Thnx for your info. Sorry but I’am a newbie and still struggling with the wiring.

 

The E07-M1101D-SMA has eight pins:

 

1 GND

2 VCC

3 GDO0

4 CSN

5 SCK

6 MOSI

7 MISO

8 GDO2

 

Where to your info the olimex has the pins (EXT)

 

1 GND

2 +3.3V

3 UIRXD GPI36

4 UITXD GPI04

5 I2C-SDA GPIO13

6 I2C-SCL GPIO16

7 HS2_DATA0 GPIO2

8 HS2_CMD GPIO15

9 VSPIC50 GPIO5

10 HS2_CLK GPIO14

 

In the radio configuration is asked to select the  right pins for

 

I’am struggling with the wiring, which pin of the olimex to which pin of the e07-M1101S-SMA

As well as what to fill in in the radio section for:

 

SCLK

CSN

MOSI

MISO

TX

RX

 

I would appreciate it very much I you could help me with this then I’am not an electronic goerroe

 

Thnx in advance

 

Ton Ensink

 

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/resources/ESP32-POE-ISO-GPIO.png

Qualitrade, 2024-10-10:
Dear Sir,

 

I wrote a mail this morning about the radio config.

Finally after a lot of testing finally fixed the problem.

Nevertheless thanks for your support !

 

Ton Ensink

 

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/resources/ESP32-POE-ISO-GPIO.png