rstrouse/ESPSomfy-RTS #479: Device often offline

chemelli74, 2024-10-12:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

I often find the device offline; wifi connection is not re-established so I bet it somehow crash or have a memory leak or similar.

Can you suggest a way to properly investigate what happens ?

### How to reproduce it (step by step)

```bash
1. Connect to network
2. Wait a 4/5 days
```

UsefulVid, 2024-10-15:
I have the same problem since I use ESPSomfy.
Maybe related to this:
https://github.com/rstrouse/ESPSomfy-RTS/issues/453
I will try to gather some logs.

UsefulVid, 2024-10-17:
Logs collected
`Launching web server...
Creating Web MicroServices...
WiFi Mode: 0
(evt) WiFi interface ready
WiFi station mode started
Socket Server Started...
App Version:2.4.6
                 shades.cfg exists so we are using that
Reading header at 0
                   version:24 len:76 roomSize:29 roomRecs:1 shadeSize:276 shade6
                                                                               3
                                                                               5
                                                                               !
Successfully set up the radio
Enabled receive on Pin #12 Timing: 1
                                    Initializing RX Queue
Guru Meditatin Error: Core  1 panic'ed (IllegalInstruction). Exception was unha.
Memory dump at 0x40113bfc: 000000ff 2100613e fb7cf00b
`

chemelli74, 2024-10-17:
Thx for collecting the log, I bet I have a similar issue even if so far I was not yet able to gather it.

rstrouse, 2024-10-22:
Which ESP32 do you have?  This log seems to indicate that initializing the wifi chip.  Can you check your router logs to see if the connection refused to issue an IP address.  Somewhere between the time it initialized the Wifi and the network response the (~7seconds) the connection dropped but the wifi library appears to have corrupted the code segment.  Consequently, this is also the response time for the watchdog timer to reset but 7 seconds is a crazy long time for a connection to handshake with your AP.

The reset should have reconnected on startup though unless it is sending it into the same sequence for connection.

chemelli74, 2024-10-23:
@rstrouse, is there a way to retrieve logs remotely ?
I would like to capture them all the time so when something happens I may have a clue if it's the same issue as reported by @UsefulVid or not.

rstrouse, 2024-10-23:
Unfortunately at this time I do not know of a solution that does not require creating code to send the logs over http.  There are libraries that do this but they are not really compatible with the receive portion of the code and eat up way too much space.  When I get some time I may look into creating something over the socket that ties to the serial output.

chemelli74, 2024-11-28:
This is the log form today, is still working for the moment but I see 2 strange values:

```
[19:44:32]Max Heap: 102388
[19:44:32]Free Heap: 149552
[19:44:32]Min Heap: 135272
```
vs

```
Max Heap: 65524
[19:45:00]Free Heap: 152348
[19:45:00]Min Heap: 100296
```

Shouldn't Max and Min be always similar if not the same ?

Full log:
```
[19:44:32]Max Heap: 102388
[19:44:32]Free Heap: 149552
[19:44:32]Min Heap: 135272
[19:44:34]Check github for updates...
[19:44:34]Max Heap: 110580
[19:44:34]Free Heap: 155448
[19:44:34]Min Heap: 135272
[19:44:38]Internet is Available: 1853ms
[HTTPS] GET... code: 200
[HTTPS] GET... code: 200 - -1
Max Heap: 65524
[19:44:38]Free Heap: 151308
[19:44:38]Min Heap: 100296
[19:45:00]Client 1 joining room 0
Timing WebServer: 3867ms
[19:45:00]Timing WebServer: 242ms
Socket [1] Disconnected!
Socket [1] Connected from 192.168.1.72 url: /
Initializing Socket Client 1
Max Heap: 65524
[19:45:00]Free Heap: 152348
[19:45:00]Min Heap: 100296
```

rstrouse, 2024-11-28:
No those values represent different aspects of the heap.   The max heap value is the maximum contiguous allocation space without fragments.  The min heap is the lowest threshold memory reached.  And the free heap is the total available memory.

The latter two values represent all blocks of memory.  Min, max, and free are used for different purposes.  For instance, if free memory continues to steadily drop over time then this is an indication of a memory leak.  I max memory tends to drop over time then this indicates fragmentation and inefficiency on the memory use.   Keep in mind the Wi-Fi and http client libraries will allocate memory and only free it periodically to maintain performance.  This creates some level of fragmentation that should recover over time.

closed 2026-06-20 by chemelli74