rstrouse/ESPSomfy-RTS #480: Is Wemos D1 mini supported?

Miesjelangelo, 2024-10-12:
### Hardware

ESP32

### Firmware version

V2.4.6

### Application version

V2.4.6

### What happened? What did you expect to happen?

Hi,

I'm trying to install the ESPSomfy software to a Wemos D1 mini. I believe it should be supported since the hardware guide mentioned that most if not all wireless ESP32 boards should work. 

I've tried multiple boards, none of them work. When flashing something else to check of the board and tooling are working, everything goes as expected. Tested with WLED and it booted right up. 

When flashing ESPSomfy, nothing happens. I'm not seeing errors when using ESP flasher, but I can't see any logging either so it seems that the software is not or partly installed on the device making it impossible to boot up. 

I'm not able to pull any logging from the device either. 

Is this my fault or is this specific board not supported? The guide seems simple enough to follow but I might missed something. 

Let me know if any extra information is needed, thanks!

Michel

### How to reproduce it (step by step)

```bash
Flash on a wemos D1 mini and see that it won't work :)
```

rstrouse, 2024-10-14:
Please make sure you downloaded the zip file and extracted the onboard binary.  If you are flashing SomfyController.ino.esp32.bin then you are using the wrong file.  You must use the onboard file for the original onboarding process.  For the wemos this should be `SomfyController.onboard.esp32.zip`

Download the file to your computer then extract the `SomfyController.onboard.esp32.bin` file.  Use this file to initially flash the ESP32.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware

Miesjelangelo, 2024-10-14:
Hi @rstrouse, thats the one i've been testing with. And to be sure, I tested the others as well. 

When uploading the SomfyController.onboard.esp32.bin file to the ESP (tried multiple), logs are empty, LED is off and no wifi is being broadcast. Any other suggestions perhaps? 

Thanks!

rstrouse, 2024-10-15:
Is this an 8266 D1 or an ESP32.  If it is the former it will not work on the 8266 chips this is an ESP32 only implementation.  If you want a small footprint device consider the ESP32 S3.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Mini-ESP32s3-Device-Build

Miesjelangelo, 2024-10-15:
That might the issue here, wasn't aware that there are two versions of the D1 mini. Thanks for pointing that out, will order a ESP32 S3, thanks!

closed 2024-10-15 by Miesjelangelo