rstrouse/ESPSomfy-RTS #481: LilyGO T-Internet POE REBOT NOT WORKING

moshiko2312, 2024-10-16:
### Hardware

LilyGO T-Internet POE

### Firmware version

V.2.4.6

### Application version

V.2.4.6

### What happened? What did you expect to happen?

when i press   reboot command the board  going to loop and not rebooting  i need  to make restart from the POE cable 
its do from HA and ESPSOMFY RTS WEB

[image]

[image]

OR  if i change  setting  and press  save its same problem  the board do loop and stacking 

[image]

Thanks for your  job  🙏 

### How to reproduce it (step by step)

```bash
1. i trying  from the web application

2. i trying  from Home Assistant
```

rstrouse, 2024-10-22:
Please double check the documentation for your version of the board.  I have read that the Power pin may have changed to GPIO 4 for later versions.

moshiko2312, 2024-10-30:
Its working but after X Time it's stop working.
And when trying to reboot or update setting its stack.

בתאריך יום ג׳, 22 באוק׳ 2024 ב-20:52 מאת rstrouse ***@***.***

jaarsa, 2025-09-24:
Hi, and thanks for the fantastic project!

While setting up ESPSomfy-RTS on the **LilyGO T-Internet POE** board, I noticed that the current recommended pinout includes GPIO12 and GPIO15. These are bootstrapping pins on the ESP32-WROOM-32E and can cause unreliable startup behavior unless properly handled with pull-down resistors.

To improve stability and avoid boot issues, I’d like to suggest an alternative pin configuration that works reliably and avoids known pitfalls:

[image]

This configuration has been tested with the E07-M1101D transceiver and works seamlessly with the ESPSomfy-RTS firmware. It maintains full compatibility with the SmartRC-CC1101-Driver-Lib and avoids bootstrapping conflicts that can occur with GPIO12 and GPIO15.

[image]