rstrouse/ESPSomfy-RTS #485: Weird only 15 equipment max !

danwador, 2024-10-21:
### Hardware

ESP32

### Firmware version

V2.4.7

### Application version

V2.4.7

### What happened? What did you expect to happen?

I created my different rooms ( 10 ).
I paired my 15 motors  everything works fine.
As soon as I try to pair the 16th, it pairs well and works, but at the same time I have another device that has disappeared!
Normally it is possible to go up to 32 devices.
I read that there was a limit and this is the variable SOMFY_MAX_SHADES,  

### How to reproduce it (step by step)

```bash
As soon as I try to pair the 16th devices
I can't exceed 15 equipment
```

rstrouse, 2024-10-21:
There should be no problem adding 32 devices.  Can you upload a backup of your settings from the System->Firmware tab.

danwador, 2024-10-22:
It's weird..... Until now I used the RFLINK solution but limited to 16 devices. It is for this reason that I wanted to try your solution.
Otherwise it doesn't matter I combine the 2.
[ESPSomfyRTS 2024-10-22T08_24_14.backup.pdf](https://github.com/user-attachments/files/17471326/ESPSomfyRTS.2024-10-22T08_24_14.backup.pdf)

rstrouse, 2024-10-22:
I have a several devices with 32 motors defined on it.  Please add a .txt extension to the file and upload it here or copy the contents and paste it into message.   The PDF will not allow me to restore to a device.

danwador, 2024-10-22:
[ESPSomfyRTS 2024-10-22T18_26_17.backup.txt](https://github.com/user-attachments/files/17479390/ESPSomfyRTS.2024-10-22T18_26_17.backup.txt)

danwador, 2024-10-22:
Thank you for your time

rstrouse, 2024-10-22:
I restored your backup to a device and then added another 5 shades to it without any problem.  So I have some questions.

1. Which ESP32 do you have?
2. What are you communicating with over MQTT?

danwador, 2024-10-22:
It's a ESP32-WROOM 
For MQTT I just configured it with the MQTT server information but I did not activate it.
I'm going to delete it and try again.

danwador, 2024-10-22:
After deleting information from the MQTT server, I can know add a 16 shade ! 
A big thanks rstrouse for all your work and many thanks for your help.

Maybe there was a link with my RFLINK somfy solution (16 shades), which still works on Home assistant & MQTT.

[logs.txt](https://github.com/user-attachments/files/17482347/logs.txt)

danwador, 2024-10-23:
I also added my connected Somfy sockets.
It's perfect

Everything works perfectly in home assistant

Thank you so much Rtrouse

closed 2024-10-23 by danwador