rstrouse/ESPSomfy-RTS #487: Revamp the Github Repo Pages

Kaotic3, 2024-10-26:
### Describe you new feature you'd like

I want you to understand that I am not criticising - you have made a cool thing - but if you are anything like me, and almost everyone on GitHub is - then making a thing isn't the cool bit - having people use the thing, that is the cool bit.  To know that your work is being used by people and enjoyed, is awesome - that is how I feel.

Please take this as a friendly post.

The pages to get people involved, are verbose and long.  To give you an example, I copied the Wiki page for sourcing hardware to a Word document - it was 17-pages and 3,100+ words.

I went through, and I tore it down to what it needed to be - and it is 3 pages and 500 words.

This isn't to say that the rest of it is not of value - but they can be in linked pages, that people can explore if they want to explore.

For example, under "Sourcing an ESP32" I put this information:

> The board must have at least 4Mb of flash and at least 5 pins that can be used as input or output and one pin that can be input only.
> 
> This is an example compatible Ethernet Board:
> •	Wireless Tag WT32-ETH01
> •	https://www.amazon.co.uk/Nagoyuki-WT32-ETH01-Embedded-Networking-Bluetooth-Black/dp/B0CHNT69NL/
> https://www.amazon.com/EC-Buying-WT32-ETH01-Development-Bluetooth/dp/B09Z298QJQ
> 
> This is an example compatible Wireless Board:
> **[Give Examples]**
> 
> IMPORTANT: Always verify the settings for the board are correct. While I have pulled the information for the board settings from currently available information, manufacturers can change the spec sheets so incorrect Ethernet settings can actually destroy and ESP32.
> 
> This guide will reflect both the WT32-ETH01 and the Wireless Example, please see here for alternative boards and pin information: [Link]
> 
> Picking a CC1101 Transceiver
> 
> This an example of a compatible CC1101 Transceiver:
> https://www.amazon.co.uk/CC1101-Long-Distance-Industrial-Wireless-Transceiver/dp/B09WKGYFPV/
> https://www.amazon.com/SHUATI-Frequency-Transceiver-Integrated-Industrial/dp/B0CQVMN5LM/
> 
> Enhanced Range
>
> To enhance the range, replace the antenna
> https://www.amazon.com/dp/B072Q7319V?psc=1&ref=ppx_yo2ov_dt_b_product_details
> https://www.amazon.co.uk/eightwood-Connector-Omni-directional-Vertical-Homematic-433MHz-Antenna-SMA/dp/B06ZZ379PK/

This replacement reduced the total word count by about 1,000 words, yet it contains all the information a person needs to get started in about 130 words.

At the moment the project is daunting partly because of the length.

I do not know how this will be received, it is meant as constructive feedback, to help get more people into using this tool - I am going to build it - just buying the parts now (hence the amazon links to hand) so grateful for that.

rstrouse, 2024-10-30:
I don't have any artistic love for the Wiki other than trying to convince folks that it really isn't that hard to accomplish.  There have been a large number of folks that do not know what a Dupont connector is much less an ESP32 that have successfully built the device and have it working.  Oddly, I see fewer questions on setup from users who do not even know where to start than those that have some working knowledge.  I can only assume that is because they read the entire wiki.

Much of what is written in the Wiki was written over time.  As new users came on board and shared their experience additional information was added to the wiki.  For instance, I do not have all the hardware described and originally it did include some links.  Unfortunately, as sellers came and went, so did the validity of the links.  There was an entire sourcing page that has since been removed due to feedback on various forums indicating the links no longer worked, folks found it a few bucks cheaper, or could get it shipped faster.  These go stale very quickly and it depends largely on what part of the world you are in.

The home page of the wiki is being reserved for a Quick start guide.  However, I don't have the kind of time required to write one.  Bear in mind that the hardware is varied and each carrier board is very different so a quick start gets very wizardy in a hurry.  This solution started out supporting only one piece of hardware then shifted to many very quickly.  Not all hardware is available in all countries.

Kaotic3, 2024-11-02:
> I don't have any artistic love for the Wiki

I will have a go at writing it as I go through the process of building it.  I already have the notes above.  

I will take into account the points you have raised here - but I think the "Quick Start" can be limited to a single version of the ESP and a simple walk-through - which is what I intend to write.

I do understand how things can get complex, but that isn't for a "Quick Start" you have pointers to the complexity that exists elsewhere.  You can have a step-by-step this is how you create it using this and this.

I will see how it goes, see what it takes, and see what I end up with - it won't be tomorrow, I work full-time this is a side project but when I have something I will provide it and you can use it, adapt it or bin it at your leisure.

Nickduino, 2024-11-13:
> The pages to get people involved, are verbose and long.

I'm the type of guys who doesn't like long introductions, instructions repeating themselves, who watches YouTube at 2X, etc....but I loved every sentence I read in the wiki, instructions are clear, exhaustive and enjoyable.

Nickduino, 2024-11-13:
> levity

There. That levity inspires self-confidence in ones ability to make things work and make it enjoyable to read all instructions.

Personally, I wouldn't change a thing.