rstrouse/ESPSomfy-RTS #488: RTS motor won't pair

wjschmitt, 2024-10-27:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

I cannot get the shade to pair.  I've triple-checked the pin connections.   The ESP32 is a ESP-WROOM-32.  The transceiver is the blue one.  I can get the frequency set and see output from the remote in the logs.  

### How to reproduce it (step by step)

```bash
1. Go to Somfy / Shades / Edit shade
2. Click on Pair Shade, press the Prog button on remote, shade jogs.
3. Press the Prog button, shade won't jog.  Tried repeated presses, and pressing and holding. No jog.
...
```

rstrouse, 2024-10-29:
Please check the wire that is connected to pin 3 on the transceiver.   Trace it back to the ESP32 and ensure that the pin it is connected to is the one that you have configured in the UI.

wjschmitt, 2024-10-30:
OK, I'm an idiot.  I was changing the pin settings around and forgot to set them back.  It is working now.  Sorry for my ignorance.

closed 2024-10-30 by wjschmitt